platformio/platformio-core #1682: Home: Could not import Arduino project

tarzan115, 2018-06-14:
`Server error`

I can not import Arduino project in vscode. please help!

ivankravets, 2018-06-15:
Could you reproduce it again?

tarzan115, 2018-06-16:
I got that error all the time I try to import. it also errors when I create a new project
[image]

ivankravets, 2018-06-16:
Please open PIO IDE Terminal and type `pio home -p 8055` and try to repeat your workflow. Do you see any errors in a console?

tarzan115, 2018-06-18:
No, I don't!
after the command, it auto open `127.0.0.1:8055`
[image]

ivankravets, 2018-06-18:
Did you try to import Arduino project?

tarzan115, 2018-06-18:
yes, I tried to import but I got that error. after that, I try to create a new project but it got the same error.

ivankravets, 2018-07-11:
Please open PIO IDE Terminal and type `pio update`. Now, it should show more explained exception report.

tarzan115, 2018-07-12:
here what I got
`[Error 2] The system cannot find the file specified: u'C:\\Users\\tarzan115\\Documents'`

ivankravets, 2018-07-12:
Which folder do you select when trying to import? `C:\\Users\\tarzan115\\Documents`?

ivankravets, 2018-07-16:
Please share a screenshot from PIO Home > Import Arduino Project wizard before pressing “Import” button.

tarzan115, 2018-07-16:
I tried to create `Documents` folder in user folder because I don't have that folder before.
and then `PIO` still show error cannot find folder `PlatformIO`, and I create folder `PlatformIO` and error continue show cannot find folder `Projects` and so on...
here is the errors
[image]

ivankravets, 2018-07-16:
You can import THE ONLY Arduino projects, where folder contains the INO file with the same name as folder name. In other cases, please use Open Project or Create New porject.

closed 2018-07-16 by ivankravets

tarzan115, 2018-07-16:
I did. I had import INO file with the same name as the folder. but PIO shows that error.

ivankravets, 2018-07-16:
Please read my previous post carefully. I asked you to show me a real Arduino project which you are going to import. Of course, and error after "Import" action.

tarzan115, 2018-08-22:
I figure out that when I move my folder "Documents" in another place and my windows user folder will not have "Documents" anymore. so now I left the "Documents" to default, that mean windows user folder will have "Documents" folder and this error gone.
error:
```
|--C:
|   |--Users
|   |   |--tarzan115 (user folder and not include "Documents")
|   |   |   |--.platformio
```

work:
```
|--C:
|   |--Users
|   |   |--tarzan115
|   |   |   |--.platformio
            |--Documents   <-- this makes everything work.
```

hopefully, this information will help somehow