platformio/platformio-core #1683: Error compiling .ino files with functions returning reference types

cjastone, 2018-06-14:
### Configuration

**Operating system**: Windows 10

**PlatformIO Version** (`platformio --version`): 3.6.0a2

### Description of problem
When building an .ino file with functions returning reference types, the file is not correctly preprocessed.  If the functions are not declared in the order in which they are called, the build fails with "function was not declared in this scope".  The same code compiles without error in the Arduino IDE.

#### Steps to Reproduce
1. Create a function that returns a reference type (see below for an example)
2. Create a function that calls the function that returns the reference type.   Ensure this second function is declared before the function that it calls
3. Build project - failure will result with "function not declared in this scope" error

### Actual Results
```
bug_report.ino: In function 'void normalFunction()':
bug_report.ino:5:35: error: 'jsonFunction' was not declared in this scope
JsonObject& json = jsonFunction();
```

### Expected Results
Project should build successfully.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:esp12e]
platform = espressif8266
board = esp12e
framework = arduino

[platformio]
src_dir = bug_report
lib_dir = ~\Documents\Arduino\libraries
```

**Source file to reproduce issue:**
```cpp
#include "Arduino.h"
#include "ArduinoJson.h"

void normalFunction() {
  JsonObject& json = jsonFunction();
}

JsonObject& jsonFunction() {
  return JsonObject::invalid();
}

void setup() {}
void loop() }{
```

### Additional info

ivankravets, 2018-06-15:
Please re-test with `pio upgrade --dev`. Thanks!

closed 2018-06-15 by ivankravets

Legion2, 2020-06-06:
This error still occurs if the `&` is aligned to the right:
```c++
#include "Arduino.h"
#include "ArduinoJson.h"

void normalFunction() {
  JsonObject &json = jsonFunction();
}

JsonObject &jsonFunction() {
  return JsonObject::invalid();
}

void setup() {}
void loop() {}
```

ivankravets, 2020-06-07:
Please use CPP instead INO if possible => https://docs.platformio.org/en/latest/faq.html#convert-arduino-file-to-c-manually

commit c055ed48 (2018-06-15): Fix preprocessor for Arduino sketch when function returns certain type // Resolve #1683
commit 19844c89 (2018-06-15): Merge branch 'hotfix/v3.5.4' into develop

* hotfix/v3.5.4:
  Fix preprocessor for Arduino sketch when function returns certain type // Resolve #1683