platformio/platformio-core #1698: Could not load boards list

digirke, 2018-06-21:
PIO Core Call Error: Error: Please specify name, url and vendor fields for C:\Users\user\.platformio\platforms\atmelavr\boards\package_MCUdude_MegaCore_index.json

ivankravets, 2018-06-21:
Duplicate of https://github.com/platformio/platformio-core/issues/1697

closed 2018-06-21 by ivankravets

marius-ne, 2021-09-24:
I have exactly the same issue: I'm trying to add a custom board for Platform IO installed via VS Code.
When I'm trying to add the board definition in "C:\Users\<name>\.platformio\boards\<board>.json":

```json {
  "my_board" : {
      "build": {
      "arduino": {
	      "ldscript": "flash_with_bootloader.ld"
      },
      "core": "adafruit",
      "cpu": "cortex-m0plus",
      "extra_flags": "-DARDUINO_TRINKET_M0 -DCRYSTALLESS -DADAFRUIT_TRINKET_M0 -DARM_MATH_CM0PLUS -D__SAMD21E18A__",
      "f_cpu": "48000000L",
      "hwids": [
	      [
	      "0x239A",
	      "0x801E"
	      ],
	      [
	      "0x239A",
	      "0x001E"
	      ]
      ],
      "mcu": "samd21e18a",
      "system": "samd",
      "usb_product": "Adafruit Trinket M0",
      "variant": "standard",
      "zephyr": {
	      "variant": "standard"
      }
      },
      "debug": {
      "jlink_device": "ATSAMD21E18",
      "openocd_chipname": "at91samd21e18",
      "openocd_target": "at91samdXX",
      "svd_path": "ATSAMD21E18A.svd"
      },
      "frameworks": [
          "arduino",
          "zephyr"
       ],
      "name": "<name>",
      "upload": {
      "disable_flushing": true,
      "maximum_ram_size": 32768,
      "maximum_size": 262144,
      "native_usb": true,
      "offset_address": "0x2000",
      "protocol": "jlink",
      "protocols": [
	      "sam-ba",
	      "blackmagic",
	      "jlink",
	      "atmel-ice"
      ],
      "require_upload_port": true,
      "use_1200bps_touch": true,
      "wait_for_upload_port": true
      },
      "url": "https://www.adafruit.com/product/3500",
      "vendor": "Adafruit"
   
  }
}
```
It's adapted from the "adafruit_trinket_m0.json" in the atmelsam platform boards. The only thing I added was the "my_board" dict at the beginning and I changed some of the values.

But when I'm trying to create a new project in PIO Home via the boards dropdown and selecting my new board "<name> (Adafruit)", I get this error:

> PIO Core Call Error: "Error: Please specify name, url and vendor fields for C:\\Users\\meneu\\.platformio\\boards\\<board>.json"

ivankravets, 2021-09-25:
Does it work if you restart VSCode?

marius-ne, 2021-09-25:
No, I tried opening and closing VS-Coden and reinstalling VS-Code / Platform-IO. No success. I should mention that I'm following this guide: [https://docs.platformio.org/en/latest/platforms/creating_board.html](url)

ivankravets, 2021-09-25:
Please attach here your board JSON

marius-ne, 2021-09-25:
For everyone else: I changed the file to the version below and put it in the same folder and it worked.

```json
{
    "build": {
    "arduino": {
        "ldscript": "flash_with_bootloader.ld"
    },
    "core": "adafruit",
    "cpu": "cortex-m0plus",
    "extra_flags": "-DARDUINO_TRINKET_M0 -DCRYSTALLESS -DADAFRUIT_TRINKET_M0 -DARM_MATH_CM0PLUS -D__SAMD21E18A__",
    "f_cpu": "48000000L",
    "hwids": [
        [
        "0x239A",
        "0x801E"
        ],
        [
        "0x239A",
        "0x001E"
        ]
    ],
    "mcu": "samd21e18a",
    "system": "samd",
    "usb_product": "Adafruit Trinket M0",
    "variant": "FirstPrototype",
    "zephyr": { 		
        "variant": "FirstPrototype"
    }
    },
    "url": "url",
    "vendor": "vendor",
    "name": "name",
    "debug": {
    "jlink_device": "ATSAMD21E18",
    "openocd_chipname": "at91samd21e18",
    "openocd_target": "at91samdXX",
    "svd_path": "ATSAMD21E18A.svd"
    },
    "platforms": ["atmelsam"],
    "frameworks": [
        "arduino",
        "zephyr"
     ],
    "upload": {
    "disable_flushing": true,
    "maximum_ram_size": 32768,
    "maximum_size": 262144,
    "native_usb": true,
    "offset_address": "0x2000",
    "protocol": "jlink",
    "protocols": [
        "sam-ba",
        "blackmagic",
        "jlink",
        "atmel-ice"
    ],
    "require_upload_port": true,
    "use_1200bps_touch": true,
    "wait_for_upload_port": true
    } 
}
```