platformio/platformio-core #1719: platformio.ini: extra_scripts: CPPDEFINES are not honoured

devimc, 2018-07-04:
What kind of issue is this?

- [ ] Question. This issue tracker is not the place for questions. If you want to ask how to do
      something, or to understand why something isn't working the way you expect it to, use
      our Community Forums https://community.platformio.org

- [ ] PlatformIO IDE. All issues related to PlatformIO IDE should be reported to appropriate repository
      https://github.com/platformio/platformio-atom-ide/issues

- [ ] Development Platform or Board. All issues related to Development Platforms or Embedded Boards
      should be reported to appropriate repository. 
      See full list with repositories and search for "platform-xxx" repository related to your hardware
      https://github.com/platformio?query=platform-

- [ ] Feature Request. Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [x] PlatformIO Core. If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: GNU/Linux Gentoo

**PlatformIO Version** (`platformio --version`): 3.5.4

### Description of problem
After updating to the version 3.5.4, `CPPDEFINES` specified by using `extra_scripts` are not part of the command line to build the project

#### Steps to Reproduce

```
platformio run -v
```

### Actual Results

`-DSP_DEBUG=1` is not part of the command line to build the project

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:esp12e]
platform = espressif8266
extra_scripts = prebuild.py
board = esp12e
framework = arduino
```

**Source file to reproduce issue:**

*prebuild.py*

```py
Import("env")
import esp_config

env.Append(CPPDEFINES=("SP_DEBUG", esp_config.debug))
```

*esp_config.py*
```py
debug=1
```

### Additional info

Workaround

*prebuild.py*

```py
Import("env")
import esp_config

env.Append(CCFLAGS=["-DSP_DEBUG=%d" % esp_config.debug])
```

ivankravets, 2018-07-04:
See http://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#before-pre-and-after-post-actions

```
Import("env", "projenv")

# access to global build environment
print env

# access to project build environment (is used source files in "src" folder)
print projenv
```

closed 2018-07-04 by ivankravets

sarfata, 2018-07-06:
@ivankravets My project is also affected by this change. This broke my build. It's going to affect a lot of people who use extra_scripts. Why did you decide to introduce this in a minor version? Maybe in the future you could have a more graceful transition, like when you change options in `platformio.ini`.

devimc, 2018-07-06:
@ivankravets thanks for the link :+1:

ivankravets, 2018-07-08:
@sarfata sorry for the issue. Normally, we follow Major.Minor.Patch model for releases. This is the first incident in the last 4th years.  Indeed, there was a critical issue where mbed-based projects could not work correctly. Global construction environment per frameworks and dependencies brought different issues. Also, `build_unflags` option did work correctly.

I've just significantly improved docs for advanced scripting and explained about 2 different construction environments.