platformio/platformio-core #1725: Error: Unknown board ID 'max32620hsp'

amitchone, 2018-07-06:
### Configuration

**Operating system**: Windows 10

**PlatformIO Version**: 3.6.0a5

### Description of problem
Failing to build using custom platform.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:mbed_586]
platform = mbed_586
board = max32620hsp
framework = mbed
```

**The content of custom platform`platforms.json`:**
```ini
{
  "name": "mbed_586",
  "title": "mbed 5.8.6",
  "description": "Maxim's microcontrollers provide low-power, efficient, and secure solutions for challenging embedded applications. Maxim's processors embed cutting-edge technologies to secure data and intellectual property, proven analog circuitry for real-world applications, and battery-conserving low power operation.",
  "url": "https://www.maximintegrated.com/en/products/digital/microcontrollers.html",
  "homepage": "http://platformio.org/platforms/maxim32",
  "license": "Apache-2.0",
  "engines": {
    "platformio": "^3.0.0"
  },
  "repository": {
    "type": "git",
    "url": "https://github.com/platformio/platform-maxim32.git"
  },
  "version": "0.0.1",
  "packageRepositories": [
    "https://dl.bintray.com/platformio/dl-packages/manifest.json",
    "http://dl.platformio.org/packages/manifest.json"
  ],
  "frameworks": {
    "mbed": {
      "package": "framework-mbed",
      "script": "builder/frameworks/mbed/mbed.py"
    }
  },
  "packages": {
    "toolchain-gccarmnoneeabi": {
      "type": "toolchain",
      "version": ">=1.60301.0"
    },
    "framework-mbed": {
      "type": "framework",
      "optional": true,
      "version": "https://github.com/ARMmbed/mbed-os.git#mbed-os-5.8"
    },
    "tool-pyocd": {
      "type": "debugger",
      "optional": true,
      "version": "~0.801.0"
    }
  }
}

```

**Error message:**
```
Processing mbed_586 (platform: mbed_586; board: max32620hsp; framework: mbed)
--------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Error: Unknown board ID 'max32620hsp'
```

### Additional info
Has a recent PIO update withdrawn support for the MAX32620HSP in line with mbedOS withdrawing support from version 5.9?

ivankravets, 2018-07-06:
See 

* https://github.com/platformio/platform-maxim32/releases
* http://docs.platformio.org/en/latest/projectconf/section_env_general.html#platform

You can use previous version of dev/platform via `platform = maxim32@2.2.0`

ivankravets, 2018-07-06:
Please move issue to https://github.com/platformio/platform-maxim32/issues

closed 2018-07-06 by ivankravets