platformio/platformio-core #1728: Cannot pass compiler option to C++ only

andig, 2018-07-08:
What kind of issue is this?

- [x] PlatformIO Core. If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: OSX

**PlatformIO Version** (`platformio --version`): 3.5.4

### Description of problem

I'm receiving compiler warnings about member initialization order. To fix I'd like to add `-Wno-reorder` build option.

#### Steps to Reproduce

1. add `build_flags=-Wno-reorder` to `pio.ini`- results in 

        cc1: warning: command line option '-Wno-reorder' is valid for C++/ObjC++ but not for C [enabled by default]

2. add `build_flags=-Wp,no-reorder` to `pio.ini` as per docs this should be for CPP only- results in 

        cc1plus: fatal error: no-reorder: No such file or directory

### Actual Results

See above

### Expected Results

At least 2) should pass option to cpp compiler and run without error.

andig, 2018-07-08:
The directory error in case 2) is still confusing according to http://docs.platformio.org/en/latest/projectconf/section_env_build.html. Or is this only supported on 3.6?

ivankravets, 2018-07-08:
> The directory error in case 2) 

https://gcc.gnu.org/onlinedocs/gcc-5.2.0/gcc/Preprocessor-Options.html

>> You can use -Wp,option to bypass the compiler driver and pass option directly through to the preprocessor. 

preprocessor != CXXFLAGS

-------------

Maybe, later we can build this DB and do better flags parsing. Nevertheless, I can guarantee any ETA.

Please use http://docs.platformio.org/en/latest/projectconf/advanced_scripting.html

```
Import("projenv")
projenv.Append(CXXFLAGS=["-Wno-reorder"])
```
If you need to pass this flag to global construction environment, please use `Import("env")` and `env.` later. All these things are documented.

Does it work now?

andig, 2018-07-09:
> preprocessor != CXXFLAGS

Another thing learned.

> Does it work now?

Sigh ;) not quite, although I understand where you're getting at. My `build-helper.py` is exactly your script but doesn't work:

    *** Import of non-existent variable ''projenv''

I've made sure scons is installed:

	❯ scons

	scons: *** No SConstruct file found.
	File "/usr/local/Cellar/scons/3.0.1/libexec/scons-local/SCons/Script/Main.py", line 924, in _main

> Maybe, later we can build this DB and do better flags parsing. Nevertheless, I can guarantee any ETA.

That sounds like a lot of effort and probably error prone as new flags seem to be coming out time and again. Really just an odd shot, but wouldn't something like `-Wcxx,no-reorder` be enough to pass any `cxx`-prefixed option to the `CXXFLAGS`?

ivankravets, 2018-07-09:
PIO uses own built-in SCons. 

You use old PIO Core. Please `pio upgrade`.

andig, 2018-07-09:
> You use old PIO Core. Please pio upgrade.

At the danger of being stupid:

    ❯ pio upgrade
    You're up-to-date!
    PlatformIO 3.5.4 is currently the newest version available.

...when run in shell or in vscode's terminal.

ivankravets, 2018-07-09:
Please share content of platformio.ini and extra script

andig, 2018-07-09:
Verbatim:

    [platformio]
    env_default=esp8266
    #env_default=esp32

    [common_env_data]
    build_flags=-Wno-reorder
    lib_deps=
      ESP Async WebServer@^1.1
      OneWire@^2.3
      Adafruit Unified Sensor@^1.0
      DHT sensor library@^1.3
      DallasTemperature@^3.7
      ArduinoJson@^5.1

    [env:esp8266]
    #platform=espressif8266
    platform=https://github.com/platformio/platform-espressif8266.git#feature/stage
    board=esp12e
    framework=arduino
    # lib_compat_mode=1 allows non-git versions of DHT and ESPAsyncWebServer@^1.1
    lib_compat_mode=light
    lib_ldf_mode=deep
    build_flags=-Tesp8266.flash.4m1m.ld
    extra_scripts = pre:build-helper.py
    upload_port=vzero-edd834.local
    #targets=upload
    lib_deps=
      ${common_env_data.lib_deps}
      ESPAsyncTCP@^1.1.2

    [env:esp32]
    # use stage version for added SPIFFS support
    #platform=espressif32
    platform=https://github.com/platformio/platform-espressif32.git#feature/stage
    board=featheresp32
    framework=arduino
    # lib_compat_mode=2 to avoid including ESPAsyncTCP
    lib_compat_mode=strict
    lib_ldf_mode=deep
    lib_deps=
      ${common_env_data.lib_deps}
    #  DNSServer@^1.1
      AsyncTCP@^1.0

and `build-helper.py`:

    Import("projenv")
    projenv.Append(CXXFLAGS=["-Wno-reorder"])

ivankravets, 2018-07-09:
"projenv" accessible only for "post" scripts.

andig, 2018-07-10:
Perfectly working now- sorry if I didn't catch this from the docs.

MacDada, 2023-04-30:
I'll argue that scripting is not a solution, but a workaround.

There should be an easy way to distinguish between options for C and C++.

For example, there could be `build_flags_c` and `build_flags_cxx` options.

MacDada, 2023-04-30:
https://github.com/platformio/platformio-core/issues/2144

MacDada, 2023-04-30:
I've created a similar script for now, as a _temporary_ solution/workaround: https://github.com/MacDada/DnWiFiDoorLock/commit/5533426e6a3e2b36b690b7ae90310587715b989a