platformio/platformio-core #1737: Help wanted - PlatformIO will not render correctly in Atom environment

Mikeinnc, 2018-07-14:
- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

Windows7 Ultimate 64 bit

**PlatformIO Version** v3.6.0a7

### Description of problem

 Downloaded and installed Atom v1.28.2x64 with no problems. Installed Clang as requested and following instructions. Then added package Platformio from 'Install packages' tab. Package (v3.6.0a7) appeared to install correctly. Restarted Atm and opened PlatformIO home. Plainly obvious that PlatformIO is NOT rendering correctly in Atom - see attached image:

[image]

Have tried reinstalling PlatformIO; reinstalling Atom and PlatformIO - always the same problem! Unable to do anything further (although Atom appears to be OK?). Any help or advice gratefully received!

Mikeinnc, 2018-07-14:
Also tried uninstalling 64 bit version of Atom and reinstalling 32 bit version. Then reinstalled PlatformIO - exactly the same rendering problem, which makes PlatformIO totally unusable.

ivankravets, 2018-07-14:
Please try to remove `%HOME_DIR%/.platformio/packages` folder and restart IDE

Mikeinnc, 2018-07-15:
Thanks for your advice. I did **exactly** as you suggested - got a dialog telling me PlatformIO core was 'downloading' and to be 'patient while PlatformIO installed' - which I was.....and have exactly the same problem! Closed machine down; rebooted and still the same issue. Everything else in Atom appears to work correctly - text renders correctly. Just PlatformIO is totally unusable. Any other advice is welcome!! Thanks - Mike

ivankravets, 2018-07-15:
Please open PIO IDE Terminal and type `pio home`. Does it work in your browser?

Mikeinnc, 2018-07-17:
Hello Ivan - thanks for the advice. Unfortunately, the answer is no - it doesn't render correctly in a browser. I tried Chrome; Firefox; Opera; Vivaldi and SeaMonkey and they all look the same as the picture I sent. IEv11 was interesting - nothing showed at all! Blank screen! (That's leaving Atom open with the terminal command having run and the line 'PlatformIO Plus (https://pioplus.com) v1.4.0' and a 'little house' showing - and putting 127.0.0.1:8008 in URL field). But - I never use IE so maybe that's a red herring? One thing - the first time I used the terminal command, a red dialog appeared in the top rh corner telling me to 'remove /.platformio/packages/tool-pioplus/pioplus.exe'. I had to close Atom, of course, to do that, but it hasn't reappeared and doesn't seem to have made any difference? Unfortunately, I didn't grab a pic of the dialog. This is strange - it works fine on my laptop which is also Win 7.

ivankravets, 2018-07-17:
We have just released new PIO Core+ and PIO Home. Please open IDE Terminal and type
```
pio upgrade --dev
pio update
```

Next. If you still have this issue, please open this URL in Chrome, then "Chrome > More tools > Developer Tools > Console". Do you see any errors?

Mikeinnc, 2018-07-18:
Just as an aside, I have just installed a new instance of Atom and
PlatformIO on a Windows10 computer, and it is working perfectly! There must
be an issue with my other computer that transcends all browsers - I now
have it on 3 computers and it works on 2! Hope this helps? Regards - Mike

ivankravets, 2018-07-18:
Please open `%USER_HOME%/.platformio/packages/contrib-piohome/index.html` file and remove line 38 with `link.rel = 'stylesheet';`. Close IDE. Restart. Does it work now?

Mikeinnc, 2018-07-20:
Thanks again, Ivan. Tried that - but unfortunately, still the same problem.
However, when I tried to open it in Chrome, with console active, I didn't
get an error message - just the same 'non-rendered' page. This is really
weird - I don't have any problems with other pages in browsers?

Thanks for your efforts, though - really appreciated!

Mike

ivankravets, 2018-07-26:
We've deployed new PIO Home recently. Could you try `pio update` in PlatformIO IDE Terminal and restart IDE? Did it help?

ivankravets, 2018-07-30:
Do you have antivirus software in a system? Maybe, some tools block local requests.

CumHoleZH, 2018-08-05:
Duplicate of #

derasd, 2018-08-10:
Same problem here. No AV.

ivankravets, 2018-08-10:
@derasd what is your OS?

derasd, 2018-08-11:
Win 7 Professional

Mikeinnc, 2018-08-11:
As is mine. AV is Sophos Professional. Used on all machines running Windows with no problems. Such a weird problem!

ivankravets, 2018-08-12:
Sorry, could someone explain me which software causes this issue? Thanks!

rkrasin, 2018-08-13:
Hi @Mikeinnc,

can you please try to run Atom with ```--disable-gpu``` flag and check if that fixes the issue?

Mikeinnc, 2018-08-13:
Still the same, I'm afraid! I also completely removed every trace of Atom I
could find on the machine, and reinstalled a new instance of v1.29 64bit.
Sorry to say - same issue! PIO just will not render - in Atom or in a
browser. Thanks for your ongoing help!!

Mike

ivankravets, 2018-08-13:
Could someone provide me a TeamViewer session and send ID/Pass to me@ikravets.com? I'll try to debug this issue. Thanks!

SylvainVanIniitu, 2018-09-23:
it seems to be the same issue i had ( [https://community.platformio.org/t/error-in-the-display-of-platformio-home/3678/12](https://community.platformio.org/t/error-in-the-display-of-platformio-home/3678/12) ) and couldnt resolve even after uninstalling/reinstalling all.   i'm also on Windows 7.

Mikeinnc, 2018-12-21:
I realise that this issue has probably dropped off the radar - and to be honest, as I have an alternative - working - Atom instance on another computer, I sort-of gave up on this. However, in the last couple of weeks, I have had to completely rebuild the computer that had this issue after a major MB failure. New motherboard (well, s/h really and carefully removed from my wife's computer!!); new video card (Nvidia vs AMD) but original os - obviously modified for the new components. As I had always assumed that the previous video system _may_ have been responsible for the issue, after I got the system working and stable, I thought I'd try Atom and PlatformIO again. Seems I was horribly wrong - still exactly the same problem with PIO just not rendering correctly at all. So, I guess I can discard the video subsystem as the possible cause of the problem! Any further comments since the last? It would, I admit be good to see this work as this computer is fast and has plenty of memory and disc space. Or is it a forlorn hope........? Thanks as always! - Mike
[image]

fmalekpour, 2019-05-09:
Issue still exists, I managed a workaround by editing .platformio\packages\contrib-pysite\twisted\web\static.py, around line 150, defined the mimetype of css, so it will be something like this:

`mimetypes.types_map.update({
	'.conf':  'text/plain',
	'.diff':  'text/plain',
	'.flac':  'audio/x-flac',
	'.java':  'text/plain',
	'.oz':    'text/x-oz',
	'.swf':   'application/x-shockwave-flash',
	'.wml':   'text/vnd.wap.wml',
	'.xul':   'application/vnd.mozilla.xul+xml',
	'.patch': 'text/plain',
	'.css': 'text/css'
	}
)
`

ivankravets, 2019-05-09:
Could you open your PIO IDE Terminal and type
```
python -c "import mimetypes; mimetypes.init(); print mimetypes.types_map"
```

Do you see `'.css': 'text/css'` in this dictionary?

fmalekpour, 2019-05-09:
I did, here is the result:

> {'.obj': 'application/octet-stream', '.zxp': 'application/x-mmxp', '.f4v': 'video/mp4', '.mrw': 'image/MRW', '.raw': 'image/RAW', '.wsc': 'text/scriptlet', '.ras': 'image/x-cmu-raster', '.ram': 'application/x-pn-realaudio', '.3gpp': 'video/3gpp', '.raf': 'image/RAF', '.fcdt': 'application/vnd.adobe.formscentral.fcdt', '.website': 'application/x-mswebsite', '.bmp': 'image/bmp', '.dvi': 'application/x-dvi', '.aif': 'audio/aiff', '.pdx': 'application/vnd.adobe.pdx', '.adts': 'audio/vnd.dlna.adts', '.p10': 'application/pkcs10', '.csv': 'application/vnd.ms-excel', '.p12': 'application/x-pkcs12', '.css': 'text/plain', '.csh': 'application/x-csh', '.pdf': 'application/pdf', '.pl': 'text/plain', '.htc': 'text/x-component', '.hta': 'application/hta', '.py': 'text/plain', '.xml': 'text/xml', '.ps': 'application/postscript', '.htm': 'text/html', '.fif': 'application/fractals', '.tsv': 'text/tab-separated-values', '.xlm': 'application/vnd.ms-excel', '.au': 'audio/basic', '.sit': 'application/x-stuffit', '.cat': 'application/vnd.ms-pki.seccat', '.mp4v': 'video/mp4', '.ief': 'image/ief', '.texinfo': 'application/x-texinfo', '.xaml': 'application/xaml+xml', '.wax': 'audio/x-ms-wax', '.ec3': 'audio/ec3', '.exe': 'application/x-msdownload', '.library-ms': 'application/windows-library+xml', '.rat': 'application/rat-file', '.ade': 'application/msaccess', '.vssx': 'application/vnd.ms-visio.viewer', '.vssm': 'application/vnd.ms-visio.viewer', '.adp': 'application/msaccess', '.mpeg': 'video/mpeg', '.adt': 'audio/vnd.dlna.adts', '.sgml': 'text/x-sgml', '.spc': 'application/x-pkcs7-certificates', '.spl': 'application/futuresplash', '.bat': 'text/plain', '.emf': 'image/x-emf', '.eml': 'message/rfc822', '.bcpio': 'application/x-bcpio', '.m2t': 'video/vnd.dlna.mpeg-tts', '.m2v': 'video/mpeg', '.arw': 'image/ARW', '.wav': 'audio/wav', '.mxp': 'application/x-mmxp', '.xbm': 'image/x-xbitmap', '.txt': 'text/plain', '.jxr': 'image/vnd.ms-photo', '.ts': 'video/vnd.dlna.mpeg-tts', '.tr': 'application/x-troff', '.dng': 'image/DNG', '.dotx': 'application/vnd.openxmlformats-officedocument.wordprocessingml.template', '.dotm': 'application/vnd.ms-word.template.macroEnabled.12', '.ols': 'application/vnd.ms-publisher', '.3gp2': 'video/3gpp2', '.etx': 'text/x-setext', '.movie': 'video/x-sgi-movie', '.appcontent-ms': 'application/windows-appcontent+xml', '.png': 'image/png', '.mhtml': 'message/rfc822', '.tar': 'application/x-tar', '.nrw': 'image/NRW', '.pns': 'image/pns', '.wvx': 'video/x-ms-wvx', '.pef': 'image/PEF', '.cer': 'application/x-x509-ca-cert', '.wsdl': 'application/xml', '.xesc': 'video/xesc', '.crw': 'image/CRW', '.rw2': 'image/RW2', '.pgm': 'image/x-portable-graymap', '.vsdx': 'application/vnd.ms-visio.viewer', '.tts': 'video/vnd.dlna.mpeg-tts', '.crl': 'application/pkix-crl', '.vsdm': 'application/vnd.ms-visio.viewer', '.m1v': 'video/mpeg', '.lpcm': 'audio/l16', '.avi': 'video/avi', '.wdp': 'image/vnd.ms-photo', '.aaui': 'application/vnd.adobe.acrobat.aaui+xml', '.mht': 'message/rfc822', '.wma': 'audio/x-ms-wma', '.wmf': 'image/x-wmf', '.cr2': 'image/CR2', '.wmz': 'application/x-ms-wmz', '.wmx': 'video/x-ms-wmx', '.pwz': 'application/vnd.ms-powerpoint', '.ics': 'text/calendar', '.nef': 'image/NEF', '.xlsm': 'application/vnd.ms-excel.sheet.macroEnabled.12', '.wmv': 'video/x-ms-wmv', '.ac3': 'audio/vnd.dolby.dd-raw', '.jsxinc': 'text/plain', '.php5': 'text/plain', '.php4': 'text/plain', '.jpeg': 'image/jpeg', '.gtar': 'application/x-gtar', '.p7m': 'application/pkcs7-mime', '.nws': 'message/rfc822', '.p7c': 'application/pkcs7-mime', '.p7b': 'application/x-pkcs7-certificates', '.mts': 'video/vnd.dlna.mpeg-tts', '.der': 'application/x-x509-ca-cert', '.p7s': 'application/pkcs7-signature', '.p7r': 'application/x-pkcs7-certreqresp', '.hxk': 'application/xml', '.eps': 'application/postscript', '.adobebridge': 'application/x-bridge-url', '.hxd': 'application/octet-stream', '.tex': 'application/x-tex', '.osdx': 'application/opensearchdescription+xml', '.rtx': 'text/richtext', '.sst': 'application/vnd.ms-pki.certstore', '.hxa': 'application/xml', '.fdf': 'application/vnd.fdf', '.gif': 'image/gif', '.mod': 'video/mpeg', '.rtf': 'application/msword', '.scss': 'text/plain', '.mov': 'video/quicktime', '.xhtml': 'application/xhtml+xml', '.vdw': 'application/vnd.ms-visio.viewer', '.qt': 'video/quicktime', '.vdx': 'application/vnd.ms-visio.viewer', '.sv4cpio': 'application/x-sv4cpio', '.ppdf': 'application/vnd.ppdf', '.mpv2': 'video/mpeg', '.wiz': 'application/msword', '.vcf': 'text/x-vcard', '.json': 'application/json', '.tpl': 'text/plain', '.pnm': 'image/x-portable-anymap', '.midi': 'audio/mid', '.tiff': 'image/tiff', '.texi': 'application/x-texinfo', '.oda': 'application/oda', '.ustar': 'application/x-ustar', '.odc': 'text/x-ms-odc', '.odt': 'application/vnd.oasis.opendocument.text', '.3g2': 'video/3gpp2', '.odp': 'application/vnd.oasis.opendocument.presentation', '.ods': 'application/vnd.oasis.opendocument.spreadsheet', '.tod': 'video/mpeg', '.mpo': 'image/mpo', '.mpa': 'audio/mpeg', '.mpg': 'video/mpeg', '.mpe': 'video/mpeg', '.pot': 'application/vnd.ms-powerpoint', '.iqy': 'text/x-ms-iqy', '.ppsx': 'application/vnd.openxmlformats-officedocument.presentationml.slideshow', '.3gp': 'video/3gpp', '.ppsm': 'application/vnd.ms-powerpoint.slideshow.macroEnabled.12', '.webm': 'video/webm', '.weba': 'audio/webm', '.pko': 'application/vnd.ms-pki.pko', '.pfx': 'application/x-pkcs12', '.m4a': 'audio/mp4', '.jnlp': 'app [... cut ...] WMD': 'application/x-ms-wmd', '.docx': 'application/vnd.openxmlformats-officedocument.wordprocessingml.document', '.sv4crc': 'application/x-sv4crc', '.gitmodules': 'text/plain', '.kdc': 'image/KDC', '.mid': 'audio/mid', '.mif': 'application/x-mif', '.xht': 'application/xhtml+xml', '.dib': 'image/bmp', '.m3u': 'audio/x-mpegurl', '.wm': 'video/x-ms-wm', '.searchConnector-ms': 'application/windows-search-connector+xml', '.flac': 'audio/x-flac', '.snd': 'audio/basic'}

Note this
'.css': 'text/plain'

ivankravets, 2019-05-09:
I see. What is your Python version? Is it an official installation? What is your OS?

So, I did patching of mimetypes on our side. Could you open PlatformIO IDE Terminal and type `pio update`. Close IDE. Start IDE.

Does it work now without your hacks?

fmalekpour, 2019-05-09:
It's Python 2.7.13. Official install on VSCode, OS is the latest Windows 10 x64.

I did the update, removed my workaround and confirmed that it works properly now.

ivankravets, 2019-05-09:
@fmalekpour thank you so much for your investigation and a hint. This is why I love our community ❤️ 

Happy coding with PlatformIO!

closed 2019-05-09 by ivankravets

SylvainVanIniitu, 2019-05-09:
Yes, fixed here as well, thanks!!!