platformio/platformio-core #1738: Home: Could not initialize project

lanmushan, 2018-07-14:
PIO Core Call Error: The next files/directories have been created in C:\Users\Administrator\Documents\PlatformIO\Projects\test
platformio.ini - Project Configuration File
src - Put your source files here
lib - Put here project specific (private) libraries

Error: 
Processing genericSTM32F103ZC (platform: ststm32; board: genericSTM32F103ZC; framework: stm32cube)

Verbose mode can be enabled via `-v, --verbose` option
Error: Unknown board ID 'genericSTM32F103ZC'
 [ERROR] Took 0.91 seconds 

 [SUMMARY] 
Environment genericSTM32F103RC	[SKIP]
Environment genericSTM32F103ZC	[ERROR]
 [ERROR] Took 0.91 seconds

lanmushan, 2018-07-14:
111

ivankravets, 2018-07-14:
Please temporary switch to upstream version where this board is available

closed 2018-07-14 by ivankravets