platformio/platformio-core #1746: Home: Could not load mDNS services

jugendhacker, 2018-07-15:
Got this Error while searching for devices via mDNS:
PIO Core Call Error: Extra data: line 3 column 1 - line 3 column 124 (char 4 - 127)

ivankravets, 2018-07-15:
What is your OS?

ivankravets, 2018-07-15:
Duplicate of https://github.com/platformio/platformio-core/issues/1311

ivankravets, 2018-07-15:
Please provide an output from a system Terminal
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

closed 2018-07-15 by ivankravets

jugendhacker, 2018-07-15:
I'm using Windwo 10!

ivankravets, 2018-07-15:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

jugendhacker, 2018-07-15:
```
C:\>echo %PATH%
C:\tools\ruby25\bin;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\tools\ruby23\bin;C:\ProgramData\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDO
WS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\ATI Technologies\ATI.ACE\Core-Static;C:\Program Files (x86)\Windows Live\Shared;C:\Program Files (x86)\PuTTY\;C:\Program Files\OpenVPN\bin
;C:\ProgramData\chocolatey\bin;C:\Program Files\PuTTY\;C:\Users\Julian\AppData\Roaming\nvm;C:\Program Files\nodejs;C:\Program Files\nodejs\;C:\Program Files (x86)\Yarn\bin\;C:\Program Files (x86)\G
pg4win\..\GnuPG\bin;C:\Program Files\Git\cmd;C:\Program Files\Microsoft VS Code\bin;C:\Users\Julian\.cargo\bin;C:\Users\Julian\AppData\Local\Programs\Python\Python36-32\Scripts\;C:\Users\Julian\App
Data\Local\Programs\Python\Python36-32\;C:\tools\ruby25\bin;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\tools\ruby23\bin;C:\ProgramData\Oracle\Java\javapath;C:\WINDOWS\system32;C:\W
INDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\ATI Technologies\ATI.ACE\Core-Static;C:\Program Files (x86)\Windows Live\Shared;C:\Program Files (
x86)\PuTTY\;C:\Program Files\OpenVPN\bin;C:\ProgramData\chocolatey\bin;C:\Program Files\PuTTY\;C:\Users\Julian\AppData\Roaming\nvm;C:\Program Files\nodejs;C:\Program Files\nodejs\;C:\Program Files
(x86)\Yarn\bin\;C:\Program Files (x86)\Gpg4win\..\GnuPG\bin;C:\Program Files\Git\cmd;C:\Program Files\Microsoft VS Code\bin;C:\Users\Julian\AppData\Local\Programs\Python\Python35;C:\Users\Julian\Ap
pData\Local\Programs\Python\Python35\Scripts;C:\Users\Julian\AppData\Local\Microsoft\WindowsApps;C:\texlive\2017\bin\win32;C:\Users\Julian\Downloads\zint-2.6.2-win32;C:\Program Files\Docker Toolbox
;C:\Users\Julian\AppData\Roaming\nvm;C:\Program Files\nodejs;C:\Program Files (x86)\Nodist\bin;C:\Program Files (x86)\Nodist;C:\Users\Julian\AppData\Roaming\npm;C:\Program Files (x86)\Nmap;C:\Progr
am Files (x86)\Android\android-sdk\platform-tools;C:\apache-maven-3.5.3-bin\apache-maven-3.5.3\bin;C:\Users\Julian\AppData\Local\Yarn\bin;C:\Users\Julian\AppData\Local\Keybase\

C:\>echo %PYTHONPATH%
%PYTHONPATH%

C:\>python --version
Python 3.6.4

C:\>pip --version
pip 10.0.1 from c:\users\julian\appdata\local\programs\python\python36-32\lib\site-packages\pip (python 3.6)

C:\>pip list
Package       Version
------------- --------
adafruit-ampy 1.0.3
click         6.7
colorama      0.3.9
ecdsa         0.13
esptool       2.4.1
fusepy        2.0.4
logzero       1.3.1
osc-tools     0.0.3
pip           10.0.1
pyaes         1.6.1
pyasn1        0.4.2
pyserial      3.4
python-gnupg  0.4.3
python-osc    1.6.8
rsa           3.4.2
setuptools    28.8.0
Telethon      0.17.4.5
virtualenv    16.0.0
Cache entry deserialization failed, entry ignored
```

ivankravets, 2018-07-15:
Please open PIO IDE Terminal and type `pio home --port 8181`. Try to list mDNS services. Do you see any errors in Terminal?

jugendhacker, 2018-07-15:
No I don't get any errors