platformio/platformio-core #1750: dynamically add or replace dependencies by extrascript.py

trombik, 2018-07-16:
What kind of issue is this?

- [x] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [x] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.
------------------------------------------------------------------

### Configuration

**Operating system**:

N/A

**PlatformIO Version** (`platformio --version`):

`PlatformIO, version 3.5.4`

### Description of problem

When a library needs two incompatible libraries, there seems to be no way to dynamically create a dependency on one.

An example: a library that supports an I2C library, say either [I2C-Master-Library](https://github.com/DSSCircuits/I2C-Master-Library) or [TinyWireM](https://github.com/adafruit/TinyWireM). Both support `frameworks arduino` and `platforms 
atmelavr`, but `I2C-Master-Library` does not supports `attiny85`. If you list both libs in `library.json`, it causes compilation failure. The solution would be hide either one depending on `cppflags` in `extrascript.py`.

However, the documentation does not say that such trick is possible, probably because it is not.

The question is, is it possible? If not, it would be nice to have a future to add or replace dependencies in `library.json` by `extrascript.py`.

#### Steps to Reproduce

1. Create a library with the two dependencies in `library.json`
2. Build the library

### Actual Results

The build fails while compiling `I2C-Master-Library`.

### Expected Results

`platformio` provides means to add or replace dependency in `extrascript.py`.

ASL07, 2021-07-04:
Is this issue still open? Is there a workaround for this?
My library needs to add some library deps based on what board we are compiling for (depending on a compile flag)

trombik, 2021-07-06:
the devs never comment on this. observing their behaviors for a few years, i would say they are not willing to discuss when they keep silent.