platformio/platformio-core #1754: Missing include search paths in eclipse .cproject

kernkraftzwerg, 2018-07-18:
What kind of issue is this?
- [*] **PlatformIO Core**.

### Configuration

**Operating system**: Linux (not relevant for the bug)

**PlatformIO Version** (`platformio --version`): 3.6.0a8

### Description of problem
When creating an Arduino project for eclipse ide (board nanoatmega328), there are some standard
include paths missing in the generated .cproject. These are all located in "toolchain-atmelavr", the
include paths in "toolchain-arduino" are present. As a result the eclipse indexer and codan produce
many errors that are all related to standard types and symbols of the AVR toolchain (e.g. uint8_t).

#### Steps to Reproduce

1. platformio init --ide eclipse -b nanoatmega328
2. Import the project in eclipse, create some source file:
#include <Arduino.h>
uint8_t test;

3. Try to open the declaration of uint8_t (right click, Open Declaration).

### Actual Results
"The selected text can not be mapped to a symbol name"

### Expected Results
If eclipse knows the include path, it will open stdint.h and show the declaration of uint8_t.

**The content of `platformio.ini`:**
```ini
[env:nanoatmega328]
platform = atmelavr
board = nanoatmega328
framework = arduino
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>
uint8_t test;
```

ivankravets, 2018-07-18:
See 11 and 26 lines in https://github.com/platformio/platformio-core/blob/develop/platformio/ide/tpls/eclipse/.settings/language.settings.xml.tpl

Eclipse should automatically load all symbols and includes from toolchain. What is your version of Eclipse? I've just tested on a simple project and it works well on Mac:

[image]

kernkraftzwerg, 2018-07-18:
I'm using both eclipse oxygen and eclipse photon, both show this bug. But may be it is completely eclipse related, because the provider that is responsible for the standard include paths seems to crash on opening of the project. I will try this with a completely clean workspace and fresh eclipse setup.

kernkraftzwerg, 2018-07-18:
Ah, sorry! Seems that eclipse photon installation messed up my workspace, a clean workspace doesn't show this problem. Should have done this first :-) So please remove my bug report!

ivankravets, 2018-07-18:
Happy coding with PlatformIO! 😊

closed 2018-07-18 by ivankravets