platformio/platformio-core #1763: Unknown development platform (when Cyrillic home path in Windows ?)

dontsovcmc, 2018-07-22:
### Configuration
**Operating system**: Windows 7x86
**PlatformIO Version** (`platformio --version`): 3.5.4

### Description of problem
First use of Platformio. Upload program error. Try virtual environment - same error.

#### Steps to Reproduce
install platformio by pip
call 'platformio run -t upload'
got error:
```
$ platformio.exe run -t upload
Processing attiny85 (framework: arduino; platform: atmelavr; board: attiny85)

PlatformManager: Installing atmelavr
Downloading
Downloading
Unpacking
Unpacking
atmelavr @ 1.9.0 has been successfully installed!
c:\python27\lib\site-packages\platformio\managers\package.py:354: UnicodeWarning: Unicode unequal comparison failed                  to convert both arguments to Unicode - interpreting them as being unequal
  elif not pkg_id and manifest['name'] != name:
Error: Unknown development platform 'C:\Users\Виктор\.platformio\platforms\atmelavr'
```
I think the reason is cyrillic Windows user name.
I uninstall Platformio

Create virtual environment on another disk with latin path
Activate it
```
$ platformio run -t upload
Processing attiny85 (framework: arduino; platform: atmelavr; board: attiny85)

PlatformManager: Installing atmelavr
Downloading
Downloading
Unpacking
Unpacking
atmelavr @ 1.9.0 has been successfully installed!
e:\code\venv\lib\site-packages\platformio\managers\package.py:354: UnicodeWarning: Unicode unequal comparison failed to convert both arguments to Unicode - interpreting them as being unequal
  elif not pkg_id and manifest['name'] != name:
Error: Unknown development platform 'C:\Users\Виктор\.platformio\platforms\atmelavr'
(venv)
```

**The content of `platformio.ini`:**
```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

[env:attiny85]
platform = atmelavr
board = attiny85
framework = arduino
board_build.f_cpu  = 1000000L  ;частота на которой работает микроконтроллер attiny85
upload_protocol = arduino ;uno: stk500v1   ;arduino micro arduino

upload_flags = -P$UPLOAD_PORT -b$UPLOAD_SPEED
upload_speed = 19200

build_flags = -Wall
lib_deps = 
    https://github.com/jscrane/TinyDebugSerial.git

; Измените на COM порт программатора
upload_port = COM4 

; Для тестирования
build_flags = -D TEST_WATERIUS

; Для логгирования
; -D LOG_LEVEL_ERROR
; -D LOG_LEVEL_INFO
; -D LOG_LEVEL_DEBUG

#avrdude: safemode: Fuses OK (E:FF, H:DF, L:62)
```

dontsovcmc, 2018-07-24:
With english user name it works..

ivankravets, 2018-07-25:
See http://docs.platformio.org/en/latest/faq.html#unicodewarning-unicode-equal-comparison-failed

Could you temporary add `[platformio]` section to your `platformio.ini` with the next contents?
```ini
[platformio]
home_dir = C:\.platformio

[env:attiny85]
platform = atmelavr
board = attiny85
...
```

dontsovcmc, 2018-07-25:
Fantastic! Sorry for failed search in docs =)

closed 2018-07-25 by dontsovcmc

ivankravets, 2018-07-25:
I hope that switching to Python3 will fix all Unicode issues.

ivankravets, 2018-07-25:
Also, if you use PlatformIO IDE, it uses `C:\.platformio` by default for non-ASCII paths.