platformio/platformio-core #1770: Duplicated options if set via `build_flags`

puzrin, 2018-07-27:
### Configuration

**Operating system**: Ubuntu 14.04

**PlatformIO Version** (`platformio --version`): 3.6.0a10

### Description of problem

Some build flags are duplicated if defined via `build_flags` option.

#### Steps to Reproduce

1. Add this lines to `platformio.ini`

```
build_flags =
  -Werror
  --specs=rdimon.specs
```

2. Execute `pio run -v`

### Actual Results

Every flag from example passed twice to CLI line

### Expected Results

Expect each flag only once in CLI line

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[platformio]
description = Grinder speed control firmware, with BackEMF speed stabilization
env_default = genericSTM32F103C8

[env:genericSTM32F103C8]
platform = ststm32@~4.3.1
board = genericSTM32F103C8
framework = stm32cube
build_flags =
  -Werror
  --specs=rdimon.specs
  -lrdimon
  -lc
; Add thhis path for local files only, to use pio's `stm32f1xx_hal_conf.h`
; in bootstrap
src_build_flags =
  -I lib/stm32cubemx_init/Inc
lib_archive = false
lib_deps = stm32cubemx_init
debug_tool = stlink
```

ivankravets, 2018-07-28:
This is a bug in ST STM32 dev/platform. Please move this issue to https://github.com/platformio/platform-ststm32/issues and we will fix as soon as possible.

closed 2018-07-28 by ivankravets

ivankravets, 2018-07-28:
A temporary solution is to use an extra script and pass these flags directly to build environment. See http://docs.platformio.org/en/latest/projectconf/advanced_scripting.html

puzrin, 2018-07-28:
Yes, we already use this simple script:

```python
Import('env')
env.Append(LINKFLAGS=[ "--specs=rdimon.specs", "-lrdimon" ])
```

but i think fixing that bug is preferable. Filled https://github.com/platformio/platform-ststm32/issues/137

ivankravets, 2018-07-28:
Sure, we will fix it! Thanks for the report.