platformio/platformio-core #1787: ESP8266 OTA fails when --auth parameter is used.

helltp, 2018-08-08:
What kind of issue is this?

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**:  win10

**PlatformIO Version** (`platformio --version`): version 3.6.0

### Description of problem
ESP8266 OTA won't work if --auth flag provided. It was working perfectly 2 months ago.

#### Steps to Reproduce

1. add flags in plaformio.ini : upload_flags = --port=1234--auth=asdf
2. run OTA command with CLI: platformio run --target upload --upload-port 192.168.1.123

### Actual Results
Usage: espota.py [options]

espota.py: error: option --port: invalid integer value: '1234 --auth=asdf'
*** [upload] Error 2
### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:d1_mini]
platform = espressif8266
board = d1_mini
framework = arduino
upload_flags = --port=1234 --auth=asdf
lib_extra_dirs = X:\...\.platformio\packages\framework-arduinoespressif8266\libraries
```

### Additional info
Tried calling the espota.py directly, and I have succeeded. I guess the upload flags does not get converted correctly to espota.py parameters, and the espota.py fails.
The command I used:
espota.py --debug --progress -i 192.168.1.123 -p 1234 -a asdf -f X:\pathtoproject\.pioenvs\d1_mini\firmware.bin

ivankravets, 2018-08-08:
Please check documentation for upload_flags option. Each argument should be in the own line

closed 2018-08-08 by ivankravets

helltp, 2018-08-10:
Thanks for the quick answer. I have already checked the docs before opening a new issue, and it does not say anything about putting every argument in a new line. 
The corresponding part of the docs is here: [http://docs.platformio.org/en/latest/platforms/espressif8266.html#authentication-and-upload-options](url)
Since it was working fine before I thought that it should be reported.

Anyway thanks for the help!

ivankravets, 2018-08-14:
Thanks! Fixed in https://github.com/platformio/platformio-docs/commit/dafd1ef9d20a971b86e9f4f9f646efe0739dbd27