platformio/platformio-core #1793: lib_ignore'd library is still built

maxgerhardt, 2018-08-15:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

Ubuntu Budge 18.04.01 LTS

PlatformIO, version 3.6.0 (upgraded via `pio upgrade --dev`

### Description of problem

Since a recent PIO update, the following does not work anymore.

I create a project for the `nucleo_l476rg` and the latest `mbed` framework (5.9.4). I need to have custom modified version of mbedtls. The mbed framework also supplies its own mbedts under `framework-mbed/features/mbedtls`. 
I have put my custom mbed version under the `lib/mbedtls-custom` folder. I am sing the lines

```
lib_ldf_mode = off
lib_ignore = mbed-mbedtls mbedtls
lib_extra_dirs = lib/mbedtls-custom
lib_deps =
    mbedtls-custom
    mbed-rtos
```
to ignore the built-in mbedtls and include my own. Since an upgrade in either PIO version or `ststm32` platform version, this does not work anymore. Now PIO builds **both** libraries and the linker fails with duplicate definitions of functions.

E.g.

```cpp
.pioenvs/nucleo_l476rg/lib90c/libmbedtls-custom.a(platform.o): In function `mbedtls_platform_setup':
/home/max/Documents/XXXXXXXX/lib/mbedtls-custom/platform.c:316: multiple definition of `mbedtls_platform_setup'
.pioenvs/nucleo_l476rg/lib2b4/mbedtls/src/platform.o:/home/max/.platformio/packages/framework-mbed/features/mbedtls/src/platform.c:318: first defined here
.pioenvs/nucleo_l476rg/lib90c/libmbedtls-custom.a(platform.o): In function `mbedtls_platform_teardown':
/home/max/Documents/XXXXXXXX/lib/mbedtls-custom/platform.c:324: multiple definition of `mbedtls_platform_teardown'
.pioenvs/nucleo_l476rg/lib2b4/mbedtls/src/platform.o:/home/max/.platformio/packages/framework-mbed/features/mbedtls/src/platform.c:326: first defined here
```
In short, although I specific `mbed-mbedtls` in the `lib_ignore` and set the `ldf_mode` to none, the LDF still shows the ignored library.

``` 
LDF MODES: FINDER(off) COMPATIBILITY(soft)
Collected 34 compatible libraries
More details about "Library Compatibility Mode": http://docs.platformio.org/page/librarymanager/ldf.html#ldf-compat-mode
Scanning dependencies...
Dependency Graph
|-- <mbedtls-custom> (/home/max/Documents/LoRaTLSv2/lib/mbedtls-custom)
|-- <mbed-rtos> (/home/max/.platformio/packages/framework-mbed/rtos)
|-- <mbed-mbedtls> (/home/max/.platformio/packages/framework-mbed/features/mbedtls)
```

#### Steps to Reproduce

1. Create new project `pio init -b nucleo_l476rg`
2. Copy the `framework-mbed/feature/mbedtls` folder into `lib/mbedtls-custom`
3. Add `lib_X` flags as above to try to ignore built-in library and only use custom one
4. Watch compilation fail on duplicate function definitions

### Actual Results

Compilation fails in the linking phase since both version of the library are present.

### Expected Results

The mbedtls library built into mbed-os should have been ignored per the `lib_ignore` directive.

ivankravets, 2018-08-15:
https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-ignore

> The multiple library names are allowed, split them with comma+space “, “ or put each library name in a separate new line.

closed 2018-08-15 by ivankravets

maxgerhardt, 2018-08-15:
Thanks, will try.