platformio/platformio-core #1802: The debug tool does not work !

Navis-Raven, 2018-08-22:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 8.1

**PlatformIO Version** (`platformio --version`): Home 1.0.2 · Core 3.6.0

**HW/SW Configuration**: 
[env:disco_l475vg_iot01a]
platform = ststm32
board = disco_l475vg_iot01a
framework = stm32cube

### Description of problem
The debug tool does not work at all for STM32CUBEL4+disco_l475vg_iot01a, 

#### Steps to Reproduce

1. Click on debug button
[image]

2. Select PIO Debug
[image]

### Actual Results
An error message is displayed, saying that PlatformIO Pre-Debug is unfoundable

[image]

### Expected Results
The debugger should work

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:disco_l475vg_iot01a]
platform = ststm32
board = disco_l475vg_iot01a
framework = stm32cube
```

**Source file to reproduce issue:**
```cpp
main.h: 
#ifndef MAIN_H
#define MAIN_H

#include "stm32l4xx_hal.h"

#define LED_PIN                                GPIO_PIN_5
#define LED_GPIO_PORT                          GPIOA
#define LED_GPIO_CLK_ENABLE()                  __HAL_RCC_GPIOA_CLK_ENABLE()

#endif // MAIN_H

//END OF FILE

main.c:
#include "main.h"

void LED_Init();

int main(void) 
{
  HAL_Init();
  LED_Init();

  while(1)
  {
    HAL_GPIO_TogglePin(LED_GPIO_PORT, LED_PIN);
    HAL_Delay(100);
  }
}

void LED_Init() 
{
  LED_GPIO_CLK_ENABLE();
  GPIO_InitTypeDef GPIO_InitStruct;
  GPIO_InitStruct.Pin = LED_PIN;
  GPIO_InitStruct.Mode = GPIO_MODE_OUTPUT_PP;
  GPIO_InitStruct.Pull = GPIO_PULLUP;
  GPIO_InitStruct.Speed = GPIO_SPEED_HIGH;
  HAL_GPIO_Init(LED_GPIO_PORT, &GPIO_InitStruct);
}

void SysTick_Handler(void) 
{
  HAL_IncTick();
}
//END OF FILE

```

### Additional info

Navis-Raven, 2018-08-22:
To select the debug task, you can do also this (see below) and selecting one of the two options, and it will give exactly the same result:

[image]

Navis-Raven, 2018-08-22:
I've just tried the same with an Adafruit HUZZAH32 (ESP-32) card with Arduino, and it gave the same result

Navis-Raven, 2018-08-23:
Today I've tried with nRF52-DK with Arduino. Same problem

[image]

ivankravets, 2018-08-24:
Did you install ST-Link USB driver? See http://docs.platformio.org/en/latest/plus/debug-tools/stlink.htmlhttp://docs.platformio.org/en/latest/plus/debug-tools/stlink.html

Navis-Raven, 2018-08-27:
The link you gave me is dead
[image]

ivankravets, 2018-08-27:
Sorry, see updated link above.

Navis-Raven, 2018-08-27:
Thanks,
No I didn't installed the ST-LINK driver yet, I just followed the tutorials here:
http://docs.platformio.org/en/latest/tutorials/index.html#id1

and I applied the tutorial for each of my STM32 Discovery and my nRF-52-DK cardboard. 
As you can see here there is no driver required in the tutorial
[image]

Same thing with the nRF52-DK for which I had the same issue. 
[image]

When I saw that no driver is required in the tutorial I thought that platformio

Navis-Raven, 2018-08-27:
Ok
After installing ST-LINK driver (for STM32 Discovery IoT node), now it works with "PIO Debug (Skip Pre-Debug)" but not with "PIO Debug". The're is still one issue with one of the two modes

[image]

Navis-Raven, 2018-08-27:
###For nRF52-DK:

As previously said there is no indication in the initial tutorial to install any driver for nRF52-DK:
http://docs.platformio.org/en/latest/tutorials/nordicnrf52/arduino_debugging_unit_testing.html

When trying to debug without pre-debug ("PIO Debug (Skip Pre-Debug)"), it tries to run CMSIS-DAP debug tool by default. I've tried to install it, but there is one issue with it's installer here: https://github.com/ARMmbed/mbed-cli/issues/733

So I modified platformio.ini in order to work as j-link debugger by adding `debug_tool = jlink`. This is platformio.ini content :
```
[env:nrf52_dk]
platform = nordicnrf52
board = nrf52_dk
framework = arduino
debug_tool = jlink
```

After this I started debugging "PIO Debug (Skip Pre-Debug)" and it worked. I didn't even have to do these steps (described here at: https://docs.platformio.org/en/latest/plus/debug-tools/jlink.html#drivers)

[image]

So since it works well without running InstDrivers.exe, for what InstDrivers.exe is used for ? Is it necessary ?

I think that the tutorial documentation should includes that the these drivers are prerequisites or at least it should says how to modify platformio.ini in order to have j-link debugger selected directly.  

Thanks for your answers

ivankravets, 2018-08-27:
> Now when I select "PIO Debug" (with pre-debug) this is the same problem as declared for Discovery IoT Node: this window is displayed:

Fixed in https://github.com/platformio/platformio-core/commit/78c383eb683b05a16a82fc33957d1f52da525372

Please open IDE Terminal and type `pio upgrade --dev`.

Navis-Raven, 2018-08-27:
Good news ! 

Can you answer the question about the driver ? (see my previous comment)

> So since it works well without running InstDrivers.exe, for what InstDrivers.exe is used for ? Is it necessary ?

ivankravets, 2018-08-31:
InstDrivers.exe is attended for J-Link debug probe. if you installed these drives before, you don't need to install again.

Navis-Raven, 2018-08-31:
@ivankravets , hummm strange I don't think I've installed them before in this PC and it is working anyway, except if InstDrivers installs something in the Nordic dev board

Navis-Raven, 2018-08-31:
and why we can start debug session, if the drivers are not installed. this is strange.

commit abbe30ef (2018-10-29): Docs: Add info about drivers to the tutorial requirements // Resolve #1802