platformio/platformio-core #1804: stk500v1 upload broken in v3.6.0

NetForces, 2018-08-22:
- [X] **PlatformIO Core**.

### Configuration

**Operating system**: macOS 10.13.6 (High Sierra)

**PlatformIO Version** (`platformio --version`): 3.6.0

### Description of problem

Trying to upload sketch with v3.6.0 fails:

```
{8:36}[2.3.5]~/Documents/Net-Forces/source/siva_servo_ldr/arduino:master ✗ ➭ platformio --version
PlatformIO, version 3.6.0
{8:36}[2.3.5]~/Documents/Net-Forces/source/siva_servo_ldr/arduino:master ✗ ➭ platformio run -t upload
Processing attiny (framework: arduino; platform: atmelavr; board: attiny85)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Warning! `board_f_cpu` option is deprecated and will be removed in the next release! Please use `board_build.f_cpu` instead.
Warning! `board_mcu` option is deprecated and will be removed in the next release! Please use `board_build.mcu` instead.
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Atmel AVR > Generic ATTiny85
SYSTEM: ATTINY85 2MHz 512B RAM (8KB Flash)
Converting main.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 26 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SoftwareServo>
|-- <Adafruit SoftServo> 1.0.0
Compiling .pioenvs/attiny/src/main.ino.cpp.o
Compiling .pioenvs/attiny/liba67/SoftwareServo/SoftwareServo.cpp.o
Compiling .pioenvs/attiny/libbf1/Adafruit SoftServo_ID711/Adafruit_SoftServo.cpp.o
Archiving .pioenvs/attiny/libFrameworkArduinoVariant.a
Indexing .pioenvs/attiny/libFrameworkArduinoVariant.a
Compiling .pioenvs/attiny/FrameworkArduino/HardwareSerial.cpp.o
Compiling .pioenvs/attiny/FrameworkArduino/Print.cpp.o
Compiling .pioenvs/attiny/FrameworkArduino/Stream.cpp.o
Compiling .pioenvs/attiny/FrameworkArduino/TinySoftwareSerial.cpp.o
Archiving .pioenvs/attiny/libbf1/libAdafruit SoftServo_ID711.a
Archiving .pioenvs/attiny/liba67/libSoftwareServo.a
Compiling .pioenvs/attiny/FrameworkArduino/Tone.cpp.o
Indexing .pioenvs/attiny/libbf1/libAdafruit SoftServo_ID711.a
Indexing .pioenvs/attiny/liba67/libSoftwareServo.a
Compiling .pioenvs/attiny/FrameworkArduino/WInterrupts.c.o
Compiling .pioenvs/attiny/FrameworkArduino/WMath.cpp.o
Compiling .pioenvs/attiny/FrameworkArduino/WString.cpp.o
Compiling .pioenvs/attiny/FrameworkArduino/abi.cpp.o
Compiling .pioenvs/attiny/FrameworkArduino/main.cpp.o
Compiling .pioenvs/attiny/FrameworkArduino/new.cpp.o
Compiling .pioenvs/attiny/FrameworkArduino/wiring.c.o
Compiling .pioenvs/attiny/FrameworkArduino/wiring_analog.c.o
Compiling .pioenvs/attiny/FrameworkArduino/wiring_digital.c.o
Compiling .pioenvs/attiny/FrameworkArduino/wiring_pulse.c.o
Compiling .pioenvs/attiny/FrameworkArduino/wiring_shift.c.o
Archiving .pioenvs/attiny/libFrameworkArduino.a
Indexing .pioenvs/attiny/libFrameworkArduino.a
Linking .pioenvs/attiny/firmware.elf
Building .pioenvs/attiny/firmware.hex
Checking size .pioenvs/attiny/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [          ]   4.9% (used 25 bytes from 512 bytes)
PROGRAM: [=         ]  12.7% (used 1044 bytes from 8192 bytes)
Configuring upload protocol...
AVAILABLE: stk500v1
CURRENT: upload_protocol = stk500v1
Looking for upload port...
Uploading .pioenvs/attiny/firmware.hex
avrdude: ser_open(): can't open device "/dev/tty.usbmodem40121 -b19200": No such file or directory
[... 138 lines cut ...]
upload_protocol = stk500v1
upload_port = /dev/tty.usbmodem40121
upload_flags = -P$UPLOAD_PORT -b$UPLOAD_SPEED
upload_speed = 19200
lib_deps=
  https://playground.arduino.cc/uploads/ComponentLib/SoftwareServo.zip
  Adafruit SoftServo
```

**Source file to reproduce issue:**
I can not provide the full source, but was able to reproduce simply with this:
```cpp
#include <Adafruit_SoftServo.h>
Adafruit_SoftServo servo;

void setup() {

}

void loop() {
	delay(30);
}
```

### Additional info

ivankravets, 2018-08-24:
Please specify each flag in a new line. See http://docs.platformio.org/en/latest/projectconf/section_env_upload.html#upload-flags

closed 2018-08-24 by ivankravets

NetForces, 2018-08-24:
I confirm that using:

```
[env:attiny]
platform = atmelavr
framework = arduino
board = attiny85
board_mcu = attiny85
board_f_cpu = 2000000L
upload_protocol = stk500v1
upload_port = /dev/tty.usbmodem40121
upload_flags =
  -P$UPLOAD_PORT
  -b$UPLOAD_SPEED
upload_speed = 19200
lib_deps=
  https://playground.arduino.cc/uploads/ComponentLib/SoftwareServo.zip
  Adafruit SoftServo
```

Works with v3.6.0.

Thanks !

ivankravets, 2018-08-24:
Happy coding with PlatformIO! ;)