platformio/platformio-core #1805: Not all the tests results are displayed in unit testing (unity, test)

Navis-Raven, 2018-08-23:
What kind of issue is this?

- [X] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 8.1

**PlatformIO Version** (`platformio --version`): PlatformIO, version 3.6.0

### Description of problem

We can see this bug directly on the tutorial documentation here (so you can follow these steps to reproduce them) :

https://docs.platformio.org/en/latest/tutorials/nordicnrf52/arduino_debugging_unit_testing.html#writing-unit-tests

[image]

As you can see there is 11 tests that have run, and only 6 are displayed. There is also 1 failure which is not desplayed. 

So as we can see the bug is that the unit testing module doesn't display all the results, this is a bug. Besides, there is a failure, I don't know if it is a real one, but this failure is not displayed.

### Additional remark
It will be good if the test abstract shows "11 Test(s) 1 Failure(s) 10 Passed 0 Ignored" and not only "11 Tests 1 Failures 0 Ignored". Note also that "1 Failures" (instead of "1 Failure" or "1 Failure(s)") is a grammatical error.

Navis-Raven, 2018-08-23:
This issue is linked to this one : https://github.com/platformio/platformio-docs/issues/33

Navis-Raven, 2018-09-06:
this one is solved by the documentation

ivankravets, 2018-09-10:
Thanks! Docs also should be fixed.

closed 2018-09-10 by ivankravets