platformio/platformio-core #1808: Issue when trying to use master branch of esp-idf

wegunterjr, 2018-08-25:
What kind of issue is this?
Platform Manager
================
Platform esp-idf
--------
Error: Traceback (most recent call last):
  File "c:\users\newman\.platformio\penv\lib\site-packages\platformio\__main__.py", line 120, in main
    cli(None, None, None)
  File "c:\users\newman\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\newman\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\newman\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\newman\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\newman\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\newman\.platformio\penv\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\newman\.platformio\penv\lib\site-packages\platformio\commands\update.py", line 46, in cli
    ctx.invoke(cmd_platform_update, only_check=only_check)
  File "c:\users\newman\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\newman\.platformio\penv\lib\site-packages\platformio\commands\platform.py", line 385, in platform_update
    platform, only_packages=only_packages, only_check=only_check)
  File "c:\users\newman\.platformio\penv\lib\site-packages\platformio\managers\platform.py", line 121, in update
    p = PlatformFactory.newPlatform(pkg_dir)
  File "c:\users\newman\.platformio\penv\lib\site-packages\platformio\managers\platform.py", line 250, in newPlatform
    _instance = platform_cls(join(platform_dir, "platform.json"))
  File "c:\users\newman\.platformio\penv\lib\site-packages\platformio\managers\platform.py", line 460, in __init__
    self._manifest = util.load_json(manifest_path)
  File "c:\users\newman\.platformio\penv\lib\site-packages\platformio\util.py", line 180, in load_json
    with open(file_path, "r") as f:
IOError: [Errno 2] No such file or directory: 'C:\\Users\\newman\\.platformio\\platforms\\esp-idf\\platform.json'

ivankravets, 2018-08-27:
We don't support non-released software.

closed 2018-08-27 by ivankravets

wegunterjr, 2018-08-28:
Is the latest 3.0 or 3.1?

ivankravets, 2018-08-28:
The latest stable is 3.0.3, see https://github.com/espressif/esp-idf/tags

wegunterjr, 2018-08-28:
So, there are 
Platforms, Frameworks, and Boards
Platforms - is this where I get the stable 3.0.3 when I install the Espressif32?
I see ESP IDf listed in packages, but can't see any information about it.

[image]

wegunterjr, 2018-08-28:
I am using the current available in PlatformIO and getting strange errors:
```
Compiling .pioenvs\esp-wrover16\aws_iot\aws-iot-device-sdk-embedded-C\src\aws_iot_shadow_json.o
C:\Users\newman\.platformio\packages\framework-espidf\components\aws_iot\aws-iot-device-sdk-embedded-C\src\aws_iot_mqtt_client_common_internal.c: In function 'aws_iot_mqtt_internal_write_len_to_buffer':
C:\Users\newman\.platformio\packages\framework-espidf\components\aws_iot\aws-iot-device-sdk-embedded-C\src\aws_iot_mqtt_client_common_internal.c:60:2: error: 'FUNC_ENTRY' undeclared (first use in this function)
FUNC_ENTRY;

```
[image]