platformio/platformio-core #1814: Warning about some renamed cardboards

Navis-Raven, 2018-08-28:
What kind of issue is this?

- [X] **PlatformIO Core**.
      It's just a warning

This is a litte remark, there is some board that are renamed in platformIO
The cardboard Adafruit HUZZAH32 ESP-32 is renamed in PlatformIO Core and documentation, whil its equivalent Adafruit HUZZAH ESP-8266 is not renamed. 

[image]

https://www.adafruit.com/product/3405?utm_source=platformio&utm_medium=docs

We have to becareful about some boards, de cardboard DISCO-L475VG-IOT01A has also another name B-L475E-IOT01A. 

See the two links here:
https://os.mbed.com/platforms/ST-Discovery-L475E-IOT01A/
https://www.st.com/en/evaluation-tools/b-l475e-iot01a.html

At the beginning of my PlatformIO usage I did not find this board in the list.

ivankravets, 2018-10-12:
We use names provided by a framework. See https://github.com/espressif/arduino-esp32/blob/master/boards.txt#L1270

closed 2018-10-12 by ivankravets

Navis-Raven, 2018-10-13:
ok

Navis-Raven, 2018-10-13:
@ivankravets maybe you should add a warning to explain this in the documentation.