platformio/platformio-core #1816: pioplus account does not work !!!

SDS-Emb, 2018-08-29:
Hi I’m already registered for one month trial into PIO plus since 8 days, but when I ask for a ticket submission, it says that my password do not match with my email. Just like I’m not registererd whereas I’m already registered and connected to PIOPlus via VSCode IDE.

Also, I couldn't able to connect to PIO community with the same ID and Passwor.

I sent you mail with my id to 'contact@pioplus.com'  but I have no answer to this date. Please check my email in your mailbox. 

Thank you

ivankravets, 2018-08-29:
PIO Account & PlatformIO Community = ARE DIFFERENT THINGS.

PIO Account belongs to https://pioplus.com, where PlatformIO Community to https://platformio.org

Please create different accounts.

closed 2018-08-29 by ivankravets

SDS-Emb, 2018-08-30:
@ivankravets ok for PIO Community.But , what about my pioplus account ?
I can access to my PIO Account via VSCode IDE

PIO Account via VSCode
[image]

 while I can't do this via this webpage
[image]