platformio/platformio-core #1824: Library dependency by author

ivankravets, 2018-08-31:
Implement an option to declare library dependencies by a specific author using the next format:
```
@AUTHOR NAME/LIBRARY NAME@^1.2.3
```

For example, `platformio.ini`

```
[env:myenv]
board = ...
lib_deps = 
  @Paul Stoffregen/OneWire
```

cr1st1p, 2019-08-25:
Just a note to think about it (also related to #1138): ArduinoJWT library is present *twice* in the repository. 
Same author, different version numbering though. 
I tried to load it by name and version. (ArduinoJWT@~02)
Program picked the wrong library and then tried to search for the version - which it failed to find.

ivankravets, 2019-08-29:
Sorry for the delay. We will back to this issue soon. We need new data in registry ("owner" info)

ivankravets, 2020-08-12:
Resolved in https://github.com/platformio/platformio-core/commit/893ca1b328523650a9832ba2dbfaf14994d61584

closed 2020-08-12 by ivankravets