platformio/platformio-core #1826: Make PIO Unit Testing engines supports all other framworks listed in the documentation

Navis-Raven, 2018-08-31:
- [X] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

------------------------------------------------------------------

[image]

Feature Request:  Make PIO Unit Testing engines supports all other framworks listed in the documentation (link below) with zero configuration (no more complex configuration) :grinning: :hap:
https://docs.platformio.org/en/latest/frameworks/index.html

ivankravets, 2019-05-23:
It's very difficult to configure automatically `stm32cube` framework because it depends on GPIO configuration. This is why custom `test_transport` should be used. The only end developer know UART configuration for the chip.

Navis-Raven, 2019-05-23:
Bad news....

Navis-Raven, 2019-05-30:
@ivankravets could you make a tutorial in order to write a custom test transport ?

ivankravets, 2019-05-30:
- http://docs.platformio.org/en/latest/tutorials/ststm32/stm32cube_debugging_unit_testing.html
- https://github.com/platformio/platformio-examples/tree/develop/unit-testing/stm32cube

Navis-Raven, 2019-05-30:
Remember that I already tried to take the whole code of unittest_transport.h for my STM32 DISCO IoT board, but it didn't work.
At this time there is no tutorial of how create the code of the file unittest_transport.h

ivankravets, 2019-05-30:
>  but it didn't work. And I couldn't know why.

Please use https://community.platformio.org/

closed 2019-05-30 by ivankravets

Navis-Raven, 2019-05-30:
@ivankravets already done. DId you remember ?
https://community.platformio.org/t/stm32cube-unit-testing-tutorial-with-stm32-disco-l475vg-iot01a/5004

ivankravets, 2019-05-30:
Sorry, currently we are so busy. Please ping us after PIO Core 4.0 https://github.com/platformio/platformio-core/projects/2