platformio/platformio-core #1827: The presentation tabs (while downloading toolchains) switches too fast !

Navis-Raven, 2018-09-01:
- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 7, VSCode

**PlatformIO Version** (`platformio --version`): PlatformIO, version 3.6.1a1

### Description of problem
When choosing a new board, and downloading the libraries and toolchains, the tabs of the presentation displayed moves too rapidly, so we can't read them

[image]

#### Steps to Reproduce

1. Click on New Project and type a project name, select a board and a framework then click finish
2. A Window displays a presentation with four tabs while downloading the toolchains, framework etc...

### Actual Results
The tabs switch from one to the other automatically too fast, so we can't finish the reading of what's inside

### Expected Results
Please slow down this switch, let at least 15 s for each tab.

ivankravets, 2018-09-05:
Duplicate of https://github.com/platformio/platformio-core/issues/1442

P.S: Just move the mouse over slides and it will stop.

closed 2018-09-05 by ivankravets

Navis-Raven, 2018-09-06:
@ivankravets 
additionnal info: even if you click on the previous tab to return back it doesn't stay enough time and go further to the next tab.