platformio/platformio-core #1830: Mbed unit test example not compiling

Zeppelin15, 2018-09-03:
What kind of issue is this?

- [x ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:
Windows 10

**PlatformIO Version** (`platformio --version`):
PlatformIO, version 3.6.1a1

### Description of problem

I cloned the  examples repo and tried to run the mbed blink example : https://github.com/platformio/platformio-examples/tree/develop/unit-testing/mbed-blink

 I changed the board to my dev board
`board = nucleo_f767zi`

The project builds and uploads fine. 
But the test option breaks on compiling.

### Compiler ERROR:
`
 Executing task: C:\Users\username\.platformio\penv\Scripts\platformio.exe test 

PIO Plus (https://pioplus.com) v1.4.6
Verbose mode can be enabled via `-v, --verbose` option
Collected 1 items

======================================================================= [test/*] Building... (1/3) =======================================================================
Please wait...
.pioenvs/nucleo_f767zi/lib59f/utest/source/unity_handler.o: In function `utest_unity_assert_failure':
C:\users\username\.platformio\packages\framework-mbed\features\frameworks\utest\source/unity_handler.cpp:27: undefined reference to `utest::v1::Harness::raise_failure(utest::v1::failure_reason_t)'
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\nucleo_f767zi\firmware.elf] Error 1
======================================================================= [ERROR] Took 10.26 seconds =======================================================================
The terminal process terminated with exit code: 1

`

Looks to me unity is not compiling or linking.
Would try and debug it myself but judging from the disappearing    `output_export.cpp` and `tasks.json` that gets created during compilation we are discouraged from doing so. Guess because its a paid feature. 

#### Steps to Reproduce

1. Clone https://github.com/platformio/platformio-examples
2. Navigate to platformio-examples/unit-testing/mbed-blink/
3. Run pio test

### Actual Results
Build fails

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:nucleo_f767zi]
platform = ststm32
framework = mbed
board = nucleo_f767zi
```

### Additional info
I am signed into an account that has PIO Plus Professional Trial for 23 more days.

ivankravets, 2018-09-04:
You don’t need to install extra libraries. I see you have a conflicted named “utest”. Pl see remove it.

ivankravets, 2018-09-05:
Please reopen if you still need help

closed 2018-09-05 by ivankravets

medtechgeek, 2018-09-30:
I get a similar undefined reference error when I try to re-run a test suite that ran successfully with the current PIO-version of mid-July. My current PIO-version is 3.6.1a4. Any idea what might have changed?

Content of platformio.ini
```
[env:disco_l475vg_iot01a]
platform = ststm32
board = disco_l475vg_iot01a
framework = mbed
test_port = /dev/cu.usbmodem1423
debug_tool = stlink
monitor_speed = 115200
build_flags = -DPIO_FRAMEWORK_MBED_RTOS_PRESENT
```

Output from unit test build:

Executing task: platformio test <

PIO Plus (https://pioplus.com) v1.4.10
Verbose mode can be enabled via `-v, --verbose` option
Collected 1 items

============================================================================================= [test/*] Building... (1/3) =============================================================================================
Please wait...
/Users/username/.platformio/packages/framework-mbed/hal/mbed_ticker_api.c: In function 'initialize':
/Users/username/.platformio/packages/framework-mbed/hal/mbed_ticker_api.c:47:22: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
if ((1 << i) == frequency) {
^~
/Users/username/.platformio/packages/framework-mbed/platform/mbed_error.c:59:13: warning: 'print_error_report' declared 'static' but never defined [-Wunused-function]
static void print_error_report(mbed_error_ctx *ctx, const char *);
^~~~~~~~~~~~~~~~~~
/Users/username/.platformio/packages/framework-mbed/targets/TARGET_STM/TARGET_STM32L4/l4_retarget.c: In function '__wrap__sbrk':
/Users/username/.platformio/packages/framework-mbed/targets/TARGET_STM/TARGET_STM32L4/l4_retarget.c:56:22: warning: comparison between pointer and integer
if (heap_ind_new > &__mbed_krbs_start) {
^
/Users/username/.platformio/packages/framework-mbed/rtos/TARGET_CORTEX/mbed_rtx_handlers.c: In function 'osRtxErrorNotify':
/Users/username/.platformio/packages/framework-mbed/rtos/TARGET_CORTEX/mbed_rtx_handlers.c:44:18: warning: unused variable 'tid' [-Wunused-variable]
osThreadId_t tid = osThreadGetId();
^~~
.pioenvs/disco_l475vg_iot01a/libac3/utest/source/unity_handler.o: In function `utest_unity_assert_failure':
/Users/username/.platformio/packages/framework-mbed/features/frameworks/utest/source/unity_handler.cpp:27: undefined reference to `utest::v1::Harness::raise_failure(utest::v1::failure_reason_t)'
collect2: error: ld returned 1 exit status
*** [.pioenvs/disco_l475vg_iot01a/firmware.elf] Error 1
============================================================================================= [ERROR] Took 18.10 seconds =============================================================================================
The terminal process terminated with exit code: 1

ivankravets, 2018-10-03:
@medtechgeek could you try [test_build_project_src = true](http://docs.platformio.org/en/latest/projectconf/section_env_test.html#test-build-project-src)?

medtechgeek, 2018-10-03:
I get multiple definitions of main with that option set to true, and the undefined reference remains (see printout below).

================================================================================================ [test/*] Building... (1/3) ================================================================================================
Processing disco_l475vg_iot01a (framework: mbed; platform: https://github.com/platformio/platform-ststm32.git; board: disco_l475vg_iot01a)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: ST STM32 > ST DISCO-L475VG-IOT01A
SYSTEM: STM32L475VGT6 80MHz 128KB RAM (1MB Flash)
DEBUG: CURRENT(stlink) ON-BOARD(stlink) EXTERNAL(blackmagic, jlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 26 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <mbed-rtos>
|-- <mbed-mbedtls>
|-- <common>
|-- <mbed-unity>
|   |-- <mbed-utest>
|   |   |-- <mbed-greentea-client>
|   |   |   |-- <mbed-rtos>
Linking .pioenvs\disco_l475vg_iot01a\firmware.elf
.pioenvs/disco_l475vg_iot01a/src/main.o: In function `main':
c:\Users\username\Code\p01device/src/main.cpp:70: multiple definition of `main'
.pioenvs/disco_l475vg_iot01a/test/test_main.o:c:\Users\username\Code\p01device/test/test_main.cpp:58: first defined here
.pioenvs/disco_l475vg_iot01a/src/main.o:(.bss.pc+0x0): multiple definition of `pc'
.pioenvs/disco_l475vg_iot01a/test/output_export.o:(.bss.pc+0x0): first defined here
.pioenvs/disco_l475vg_iot01a/lib435/utest/source/unity_handler.o: In function `utest_unity_assert_failure':
C:\Users\username\.platformio\packages\framework-mbed\features\frameworks\utest\source/unity_handler.cpp:27: undefined reference to `utest::v1::Harness::raise_failure(utest::v1::failure_reason_t)'
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\disco_l475vg_iot01a\firmware.elf] Error 1
================================================================================================= [ERROR] Took 6.84 seconds =================================================================================================

ivankravets, 2018-10-04:
@medtechgeek could you provide a simple project to reproduce this issue?