platformio/platformio-core #1832: In some circonstances the PIO unified debuggers fails to start

Navis-Raven, 2018-09-05:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.
------------------------------------------------------------------

### Configuration

**Operating system**: Windows 8.1
**Board**: disco_l475vg_iot01a
**Framework**: ARM mbed

**PlatformIO Version** (`platformio --version`): PlatformIO, version 3.6.1a1

### Description of problem
In some circonstances the PIO unified debuggers fails to start

#### Steps to Reproduce

1. Build and upload the given code on the disco board
2. Start a debug session

### Actual Results
It fails

[image]

See the debug console:
```
Reading symbols from c:/Users/charafeddine/Documents/PlatformIO/Projects/180905-171410-mbed-blink/.pioenvs/disco_l475vg_iot01a/firmware.elf...
done.
PlatformIO Unified Debugger > http://bit.ly/pio-debug
PlatformIO: Initializing remote target...
GNU MCU Eclipse 64-bits Open On-Chip Debugger 0.10.0+dev-00392-gbe9ef0b0 (2018-01-12-15:03)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 500 kHz
adapter_nsrst_delay: 100
none separate
srst_only separate srst_nogate srst_open_drain connect_deassert_srst
Info : tcl server disabled
Info : telnet server disabled
Info : Unable to match requested speed 500 kHz, using 480 kHz
Info : Unable to match requested speed 500 kHz, using 480 kHz
Info : clock speed 480 kHz
Info : STLINK v2 JTAG v28 API v2 SWIM v18 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 3.219474
Info : stm32l4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : accepting 'gdb' connection from pipe
Info : device id = 0x10076415
Info : flash size = 1024kbytes
0x08000dbc in _write (fh=0, buffer=0x0, length=0, mode=<optimized out>) at C:\Users\charafeddine\.platformio\packages\framework-mbed\platform\mbed_retarget.cpp:619
619	                        return -1;
Not implemented stop reason (assuming exception): undefined
Temporary breakpoint 1 at 0x80037b0: file src\main.cpp, line 6.
Info : Unable to match requested speed 500 kHz, using 480 kHz
Info : Unable to match requested speed 500 kHz, using 480 kHz
adapter speed: 480 kHz
Unable to match requested speed 500 kHz, using 480 kHz
Unable to match requested speed 500 kHz, using 480 kHz
adapter speed: 480 kHz
Error: timed out while waiting for target halted
timed out while waiting for target halted
.pioinit:14: Error in sourced command file:
Remote communication error.  Target disconnected.: No such file or directory.
```

### Expected Results
It should start

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:disco_l475vg_iot01a]
platform = ststm32
board = disco_l475vg_iot01a
framework = mbed
debug_tool = stlink
```

**Source file to reproduce issue:**
main.cpp (it's the blinky example for arm mbed)
```cpp
#include "mbed.h"

DigitalOut myled(LED1);

int main() 
{
    while(1) 
    {
        myled = 1;
        wait(0.2);
        myled = 0;
        wait(0.2);
    }
}
```

### Additional info
To palliate this from this bug (waiting for a fix), the solution is to unplug then replug the disco board. 
The bug occurs only if you start a debug session few moment after a normal uploading. It doesn't occur when you start the debug session directly.

ivankravets, 2018-09-06:
I'm not sure that this is PlatformIO issue.  Could you wait 1-2 seconds after uploading? It seems that you need to give a few seconds to boot board.

Navis-Raven, 2018-09-06:
Since it recompile before each task I let more than 2 s. I have the same issue when I use STM32Cube with Blinky example

Ando38, 2018-09-13:
Hi,

Same issue for me except that the workaround doesn't work. I've succeeded debugging several time before the issue happened. (Nucleo + cube framework)

Navis-Raven, 2018-09-13:
@Ando38 Do the debug works in your case if you remove USB and plug it again ?

Ando38, 2018-09-13:
Unfortunately no, it doesn't

Navis-Raven, 2018-09-16:
For me it does work again...
and I don't know from where is the problem....
and I don't know if our problems have the same origin.

ivankravets, 2018-09-16:
Yes, this issue is not related to PlatformIO Core. This is hardware issue.

closed 2018-09-16 by ivankravets

Navis-Raven, 2018-09-16:
@ivankravets  how do you know ? it can be the interaction between PIO plus debugger and the DISCO IoT Node hardware.