platformio/platformio-core #1841: pio run -t monitor always uses first monitor_port even with multiple env's

blackketter, 2018-09-10:
**Operating system**:
macOS

**PlatformIO Version** (`platformio --version`):
3.6.0

### Description of problem
When trying to do `pio run -t monitor -e envname` the first monitor_port found in the file is always used, independent of the environment chosen.

#### Steps to Reproduce
With the below ini file, executing `pio run -t monitor -e strip` the monitor always connects to the monitor_port listed in env:string

Swapping the order of the environments in the ini file changes to the first one listed.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[platformio]
#env_default = string

[common]
libs =
  file://../NTPClient
  file://../Switch
  file://../WiFiThing
  file://../Console
  file://../Clock
  file://../Timezone
  Time
  file://../FastLED
  ESP8266Wifi
  ESP8266mDNS
  ESP8266HTTPClient
  ArduinoOTA

[env:string]
platform = espressif8266
framework = arduino
board = d1_mini
lib_deps = ${common.libs}
upload_port = razzlestring.local
monitor_port = socket://razzlestring.local:23

[env:strip]
platform = espressif8266
framework = arduino
board = d1_mini
lib_deps = ${common.libs}
upload_port = razzlestrip.local
monitor_port = socket://razzlestrip.local:23
```

ivankravets, 2018-10-12:
Thanks, fixed! Please `pio upgrade --dev`.

closed 2018-10-12 by ivankravets

blackketter, 2018-10-12:
Working great.  Thanks!

commit 69d01c4b (2018-10-12): Fix an issue when ``pio run -t monitor`` always uses first ``monitor_port`` even with multiple environments // Resolve #1841