platformio/platformio-core #1846: LibraryManager tries to download wrong library

dmaixner, 2018-09-16:
### Configuration

**Operating system**: Windows 10 64bit
**PlatformIO Version** (`platformio --version`): 3.6.1a4

### Description of problem

I have library, which is using EEPROM from builtin libs. I was able to run this project without problems several months ago. No I tried to run it with updated version of platformio and I get error.

#### Steps to Reproduce

1. try to run platformio run --target clean

this is valid for whatever target I use

### Actual Results
LibraryManager: Installing id=2284 @ EEPROM
Error: Could not find a version that satisfies the requirement 'EEPROM' for your system 'windows_amd64'

### Expected Results
No error expected, EEPROM was in builtin libs. This 2284 EEPROM is not correct EEPROM lib, which was located in ~/.platformio/packages/framework-arduinoespressif8266/libraries/EEPROM. It should be this one: https://github.com/esp8266/Arduino/blob/master/libraries/EEPROM/library.properties

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[platformio]
env_default = d1_mini

[common]
platform = espressif8266
framework = arduino
build_flags =
    -D _config_option_wifi
    -D _config_option_relay
lib_deps =
    https://github.com/dmaixner/esp8266-config/archive/master.zip

[env:d1_mini]
board = d1_mini
platform = ${common.platform}
framework = ${common.framework}
build_flags = ${common.build_flags}
lib_deps = ${common.lib_deps}
```

**Source file to reproduce issue:**
```cpp
#include "Config.h"

void setup()
{
    Serial.begin(9600);

    // print out some info data
    Config::printStats();
}

void loop()
{
    delay(60000);
}
```

### Additional info
I tried to uninstall platformio, removed all ~/.platformio. Then installed it again. Now there are no builtin libraries (it seems "framework-arduinoespressif8266" package is not installed for some reason), but I still get the same error "Error: Could not find a version that satisfies the requirement 'EEPROM' for your system 'windows_amd64'". I get this error even if I install "framework-arduinoespressif8266" package with --with-package.

dmaixner, 2018-09-17:
I have found temporary workaround:
1) use lib_deps to download libs, as specified earlier
2) remove lib_deps from platformio.ini

Now use 'clean' and immediately after it finishes, use 'build' or 'upload'. (If I don't use 'clean', there are some assembler errors thrown...)

I'm still looking forward to see some better solution to my problem.

ivankravets, 2018-09-21:
Duplicate of https://github.com/platformio/platformio-core/issues/1654

closed 2018-09-21 by ivankravets