platformio/platformio-core #1859: Home: Could not initialize project

johns-codings, 2018-09-23:
PIO Core Call Error: The next files/directories have been created in /home/john/PioProjects/esp8285-new
platformio.ini - Project Configuration File
src - Put your source files here
lib - Put here project specific (private) libraries

Error: Processing esp8285 (platform: espressif8266; board: esp8285; framework: arduino)
--------------------------------------------------------------------------------
Traceback (most recent call last):
File "/home/john/.platformio/packages/tool-scons/script/scons", line 97, in <module>
import pkg_resources
File "/home/john/.platformio/penv/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 947, in <module>
class Environment(object):
File "/home/john/.platformio/penv/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 951, in Environment
self, search_path=None, platform=get_supported_platform(),
File "/home/john/.platformio/penv/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 180, in get_supported_platform
plat = get_build_platform()
File "/home/john/.platformio/penv/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 380, in get_build_platform
from sysconfig import get_platform
File "/usr/lib/python2.7/sysconfig.py", line 10, in <module>
'stdlib': '{base}/'+sys.lib+'/python{py_version_short}',
AttributeError: 'module' object has no attribute 'lib'
========================== [ERROR] Took 0.11 seconds ==========================

ivankravets, 2018-09-26:
please remove `/home/john/.platformio/penv` and restart IDE.

closed 2018-09-26 by ivankravets

johns-codings, 2018-09-26:
Works as advertised! Thanks