platformio/platformio-core #1862: Feature Req: Silabs EFM8 Support

Si03, 2018-09-26:
- [X ] **Feature Request**.

Hi,
I see there is EFM32 support, so are there any plans to support the 8-bit range of Silabs devices?
The tools supplied by Silabs are ok (Simplicity Studio), but we would like to work in VSCode which is a more familiar environment where we can be more productive and have a workflow common to other device platforms.
Thanks,
Si.

zerog2k, 2018-11-19:
there is generic 8051 support now via platform `intel_mc51`, however no explicit support for efm8, although this can be added by anyone who wants to take this on.
I believe it would be functionally equivalent to supporting c8051 (earlier 8051 silabs series, i think they are probably same/similar cores), as requested in https://github.com/platformio/platform-intel_mcs51/issues/7

adding the basic cpu support would be easy - the challenge may lie in adding product-line specific library/sdk for efm8 and/or c8051, as well as supporting native upload (i.e. via some c2 programmer option)

EduardoOn45, 2022-10-04:
"Adding the basic cpu support would be easy - the challenge may lie in adding product-line specific library/sdk for EFM8 and/or c8051, as well as supporting native upload (i.e. via some c2 programmer option)"  

Exactly! It would be very nice.