platformio/platformio-core #1870: errors to build arduino firmware 2.0.x

fgmGitHub, 2018-10-07:
Errors found with Arduino Firmware build 2.0.x (latest version (today))

------------------------------------------------------------------

### Configuration

**Windows 10**:

**PlatformIO Version** (`18.1`):

### Description of problem

Processing megaatmega2560 (platform: atmelavr; board: megaatmega2560; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------
Error: Traceback (most recent call last):
  File "c:\users\fernando\.platformio\penv\lib\site-packages\platformio\__main__.py", line 120, in main
    cli(None, None, None)
  File "c:\users\fernando\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\fernando\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\fernando\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\fernando\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\fernando\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\fernando\.platformio\penv\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\fernando\.platformio\penv\lib\site-packages\platformio\commands\run.py", line 107, in cli
    result = (envname, ep.process())
  File "c:\users\fernando\.platformio\penv\lib\site-packages\platformio\commands\run.py", line 205, in process
    result = self._run()
  File "c:\users\fernando\.platformio\penv\lib\site-packages\platformio\commands\run.py", line 300, in _run
    self.verbose)
  File "c:\users\fernando\.platformio\penv\lib\site-packages\platformio\commands\run.py", line 323, in _autoinstall_libdeps
    ctx.invoke(cmd_lib_install, libraries=[lib], silent=not verbose)
  File "c:\users\fernando\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\fernando\.platformio\penv\lib\site-packages\click\decorators.py", line 26, in new_func
    return f(get_current_context().obj, *args, **kwargs)
  File "c:\users\fernando\.platformio\penv\lib\site-packages\platformio\commands\lib.py", line 103, in lib_install
    library, silent=silent, interactive=interactive, force=force)
  File "c:\users\fernando\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 352, in install
    force=force)
  File "c:\users\fernando\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 685, in install
    with LockFile(self.package_dir):
  File "c:\users\fernando\.platformio\penv\lib\site-packages\platformio\lockfile.py", line 102, in __enter__
    self.acquire()
  File "c:\users\fernando\.platformio\penv\lib\site-packages\platformio\lockfile.py", line 86, in acquire
    return self._lock()
  File "c:\users\fernando\.platformio\penv\lib\site-packages\platformio\lockfile.py", line 61, in _lock
    self._fp = open(self._lock_path, "w")
IOError: [Errno 13] Permission denied: u'y:\\Software\\3D_Printer\\Arduino\\Scalar\\Marlin-bugfix-2.0.06.10.2018\\Marlin-bugfix-2.0.x\\.piolibdeps.lock'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

[... 311 lines cut ...]
  Adafruit NeoPixel
  TMCStepper
  Servo(STM32F1)
  TMC26XStepper
  U8glib-HAL
  c1921b4

#
# Espressif ESP32
#
[env:esp32]
platform    = https://github.com/platformio/platform-espressif32.git#feature/stage
board       = esp32dev
framework   = arduino
upload_port = COM3
lib_ignore  =
  LiquidCrystal_I2C
  LiquidCrystal
  NewliquidCrystal
  LiquidTWI2
  TMC26XStepper
  c1921b4
src_filter  = ${common.default_src_filter} +<src/HAL/HAL_ESP32>

```

ivankravets, 2018-10-10:
> IOError: [Errno 13] Permission denied: u'y:\Software\3D_Printer\Arduino\Scalar\Marlin-bugfix-2.0.06.10.2018\Marlin-bugfix-2.0.x\.piolibdeps.lock'

Please remove `y:\Software\3D_Printer\Arduino\Scalar\Marlin-bugfix-2.0.06.10.2018\Marlin-bugfix-2.0.x\.piolibdeps.lock` and try again.

closed 2018-10-10 by ivankravets

fgmGitHub, 2018-10-11:
Thanks

Fernando G. Monteiro |
Enfermeiro Chefe do Hospital de Dia - Adultos
Vogal de Direção da EPOP
[1502106312236_image003.png]

 <http://www.sns.gov.pt/>  <http://www.sns.gov.pt/>

Closed #1870<https://github.com/platformio/platformio-core/issues/1870>.