platformio/platformio-core #1875: Support of custom targets in mbed OS

martinjaeger, 2018-10-12:
- [X] **Feature Request**.
      I am developing an open source product using mbed OS. The product is based on an STM32 MCU, but the board needs some board settings/files to be changed compared to the standard mbed configuration for the Nucleo boards. Mbed OS allows to define custom targets. It would be great if those could be supported in platformio.
------------------------------------------------------------------

Here is a description how custom targets can be defined in mbed:
https://os.mbed.com/docs/latest/tools/adding-and-configuring-targets.html
https://os.mbed.com/docs/v5.8/reference/porting-targets.html

I know that platformio uses a custom build system. But platformio also [supports custom boards](http://docs.platformio.org/en/latest/platforms/creating_board.html). Is it possible to integrate the mbed custom_targets.json based board description into the platformio custom_board folder (in the project directory) and use it without patching the mbed framework?

I can also try to support this work if you give me some hints where to start in platformio.

Thanks,
Martin

ivankravets, 2018-10-12:
Which option would you like to overwrite?

martinjaeger, 2018-10-12:
So far 3 configurations I would like to change which need custom board definitions:
- I want to change the timer TIM21 used for the us ticker to TIM22 because I need TIM21 for PWM output (STM32L073CZ).
- Want to be able to use the cheaper STM32L072CB with 128kb flash where no board is currently defined for. Basically only needs one change in the linker script to adjust memory size.
- Would like to jump into USB bootloader after reset if requested. This needs change in SystemInit(). See also here: https://github.com/ARMmbed/mbed-os/issues/7853

ivankravets, 2018-10-26:
Could you provide an example of your `targets.json`?

Geend, 2019-02-22:
I'm really interested in this feature as well. Currently three steps are necessary to use a custom board. 2 of which have to be done in the framework-mbed folder, which is not nice, because it is hard to distribute to other devs and machines. Also who knows what happens on framwork-mbed updates...

For context: I build a project specific custom board based on the  Nucleo F303RE and need some changes in the PeripheralPins.c mbed file.

Step 1: For now I just appended the config of my custom board to mbeds target.json (~/.platformio/packages/framework-mbed/targes):

`
{
...

"MYBOARD": {
        "inherits": ["FAMILY_STM32"],
        "supported_form_factors": [],
        "core": "Cortex-M4F",
        "extra_labels_add": ["STM32F3", "STM32F303xE", "STM32F303RE"],
        "config": {
            "clock_source": {
                "help": "Mask value : USE_PLL_HSE_EXTC | USE_PLL_HSE_XTAL (need HW patch) | USE_PLL_HSI",
                "value": "USE_PLL_HSE_EXTC|USE_PLL_HSI",
                "macro_name": "CLOCK_SOURCE"
            }
        },
	"overrides": { "lse_available": 0 }, 
        "detect_code": ["0745"],
        "device_has_add": [
            "ANALOGOUT",
            "CAN",
            "CRC",
            "SERIAL_ASYNCH",
            "FLASH",
            "MPU"
        ],
        "release_versions": ["2", "5"],
        "bootloader_supported": true,
        "device_name": "STM32F303RE"
    }
`
Step 2: I also copied the folder "~/.platformio/packages/framework-mbed/targets/TARGET_STM/TARGET_STM32F3/TARGET_STM32F303xE/TARGET_NUCLEO_F303RE" to ~/.platformio/packages/framework-mbed/targets/TARGET_STM/TARGET_STM32F3/TARGET_STM32F303xE/MYBOARD" and made some necessary changes for my custom board.

Step 3:  Copying " ~/.platformio/platforms/ststm32/boards/nucleo_f303re.json 
" to ~/.platformio/boards/myboard.json

I would be really nice if all those steps could be done inside a platformio project directory. Using project specific custom boards is kind of hard at the moment (or am I missing something here?)

martinjaeger, 2019-02-22:
Looks like it is now already possible to implement small changes in the target with the new [mbed build system configuration](https://docs.platformio.org/en/latest/frameworks/mbed.html#the-configuration-system) announced [here](https://community.platformio.org/t/arm-mbed-os-5-11-new-builder-with-support-for-mbed-app-json/6471).

As far as I can see, the `mbed_app.json` does not allow 

Does PlatformIO use the standard mbed-cli based build system now? If yes, supporting completely custom boards would only need a `--source /path/to/pio/project` option if a file `custom_targets.json` is found in the project root directory.

ivankravets, 2019-02-23:
We don't use `mbed-cli` tool. We use official ARM mbed OS Build API. You can check the latest 5.11 build scripts here ~/.platformio/packages/framework-mbed/platformio/platformio-build.py

Geend, 2019-02-26:
So is using the mbed custom_targets.json file possible right now?

valeros, 2019-06-21:
Hi @Geend @martinjaeger! We added initial support for custom targets. It would be great if you could test it with your configurations. Here is as [an example project](https://github.com/platformio/platform-nxplpc/tree/develop/examples/mbed-custom-target) and [doc page](http://docs.platformio.org/en/latest/frameworks/mbed.html#custom-targets). To test just update to the latest mbed package via `pio update`.

qtphan734, 2019-06-21:
@valeros @ivankravets the example you provided seem to be for custom boards with currently supported MCUs (eg. STM32 family).  I am working with a development board with an NXP MIMX8MQ6.

I followed the porting guide from mbed. This involves adding new targets for chip family and board, porting over boostrap (eg. linker file, startup file), drivers from NXP, implement HAL APIs.  All of this is working in my mbed fork with `mbed-cli`.

Can this be done in the initial support for custom target in PIO?

valeros, 2019-06-21:
Hi @qtphan734! That's an interesting question because we don't have any board based on MIMX8MQ6 MCU but I think it's feasible. At the moment how do you structure the code specific to your target?

qtphan734, 2019-06-21:
Hi @valeros,

I currently ported the MCUXpresso SDK for the MIMX8MQ6 from NXP.  The structure is below. It is very similar to other MCUXpresso MCUs supported by mbed (eg. MIMX1050RT). 
```
mbed-os
└── targets
    ├── targets.json
    └── TARGET_NXP
        └── TARGET_MCUXpresso_MCUS
            └── TARGET_MCUXpresso_MCUS
                └── TARGET_MIMX8MQ6
                    ├── TARGET_DARTMX8M
                    ├── device
                    ├── drivers
                    ├── PeripheralPins.h
                    ├── PortNames.h
                    ├── gpio_api.c
                    ├── gpio_object.h
                    ├── lp_ticker.c
                    ├── objects.h
                    ├── pinmap.c
                    ├── serial_api.c
                    └── u_tick.c
```

valeros, 2019-06-21:
@qtphan734 I see here two possible solutions (both of them imply that you already have `custom_targets.json` file).

- **Simple**. You keep the same workflow as you do now, With each new release, you will copy your target related code to our framework package (or you can create a separate platform and specify a link to your fork with mbed sources, but you'll need to add our build script as a submodule). Everything else should work as expected.
- **Portable**. You move your code to the `src` folder in PlatformIO project. In order to make it compile you need to specify basic gcc flags like the path to your include files and linker script, everything else should work out of the box. The big advantage of this approach is that your code and mbed sources become isolated of each other, therefore you can get rid of that fork you're using now and instead rely on the framework package in our repository,

And of course, for both cases, you'll need to create a custom PlatformIO board. It's not that hard. You can take any NXP board as the basis and modify it according to your MCU datasheet,

martinjaeger, 2019-06-25:
Made a first quick check, but it complains with a yellow "mbed build API internal error". Is there a way to get more information about the error? Running platformio with -v option didn't make a difference.

valeros, 2019-06-25:
@martinjaeger Unfortunately, that's the all possible info we can get.
Can you describe your project configuration?  Does it work with mbed-cli?

martinjaeger, 2019-06-25:
Alright, I'll put together a blinky example for that board.

Didn't try with mbed-cli yet, as I'm not really using it (PlatformIO is so much more comfortable).

martinjaeger, 2019-06-25:
Here is a test repository which throws above error: https://github.com/martinjaeger/platformio-custom-board-test

valeros, 2019-07-02:
@martinjaeger A few issues with your project:
1. Target-related code should be placed in the `src` folder.
2. Board name in `custom_targets.json` file should be the same as the name of PlatformIO board manifest.
I added [an example](https://github.com/platformio/platform-ststm32/tree/develop/examples/mbed-custom-target) based on your code. It compiles, but I'm not sure if it works correctly. Before testing, make sure you've updated all packages.

martinjaeger, 2019-07-03:
Thanks for your support.

I cloned your example and even deleted all `ststm32` and `framework-mbed` folders from `~/.platformio` before running `pio update` and `pio upgrade`, but still getting `mbed build API internal error`:

```
> Executing task: platformio run <

Processing custom_nucleo_l452re_mbed (platform: ststm32; board: custom_nucleo_l452re_mbed; framework: mbed)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
PlatformManager: Installing ststm32
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
ststm32 @ 5.4.3 has been successfully installed!
Uninstalling tool-openocd @ 1.1000.6:   [OK]
Uninstalling toolchain-gccarmnoneeabi @ 1.80201.181220:         [OK]
The platform 'ststm32' has been successfully installed!
The rest of packages will be installed automatically depending on your build environment.
PackageManager: Installing framework-mbed @ ~5.51203.190509
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/custom_nucleo_l452re_mbed.html
PLATFORM: ST STM32 > ST Nucleo L452RE
HARDWARE: STM32L452RET6 80MHz 64KB RAM (256KB Flash)
DEBUG: CURRENT(stlink) ON-BOARD(stlink) EXTERNAL(blackmagic, jlink)
write() takes exactly 2 arguments (3 given)
mbed build API internal error
================================================================================================= [ERROR] Took 39.85 seconds =================================================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```

valeros, 2019-07-03:
The custom target support hasn't been released yet. You can try to compile the example with the upstream version of the platform.
```
[env:your_env]
platform = https://github.com/platformio/platform-ststm32.git
framework = mbed
board = custom_nucleo_l452re_mbed
```

martinjaeger, 2019-07-03:
Ah, now it compiles and flashes, but doesn't blink. It seems that it gets stuck directly at the beginning of main, as I can set the state of the LED once and it stays at that state for ever. I'll try to find out why.

valeros, 2019-07-03:
That's probably because that custom target is composed of different L4 targets from the mbed package (just to show that it can be compiled). If you want to use it in your project, you definitely need to review all the source files (including startup, linker, etc)

closed 2019-07-18 by ivankravets

hhff, 2020-04-20:
hi @martinjaeger - did you determine why it doesn't blink?

martinjaeger, 2020-04-20:
> hi @martinjaeger - did you determine why it doesn't blink?

No, I switched to Zephyr RTOS in the meantime.

hhff, 2020-04-20:
Gotcha, thanks @martinjaeger !