platformio/platformio-core #1881: Feature Request - Option to disable 'Recent News'

xMateMCPE, 2018-10-18:
The 'Recent News' section inside the PlatformIO Home is cluttering the screen and generating useless traffic. An option to easily disable it would be appreciated.

drgeoffb, 2018-10-29:
I Endorse This Product or Service, albeit for a different reason. I wouldn't mind recent news and I'm not concerned with the prospect of 'generating useless traffic', but I do find the constantly moving carousel element to be really distracting. For my porpoises, therefore, simply hiding the section is enough.

This is not the option you're requesting, but by way of an interim solution, you can cause the recent news section to be hidden by editing the .css file corresponding to whatever theme you're using.

 On my systems (both Linux and Winders), the relevant files live in ${HOME}/.platformio/packages/contrib-piohome/themes/ 

 I have added the following to 'platformio-dark.css':

    .recent-news {
      display: none;
    }

and, boom! the 'Recent News' section is still there (in the html source) but not displayed. For me, this 'solves' the problem.

The actual source that handles this is in a package called 'contrib-piohome' that seems to be a part of PioPlus and not on github?

HTH,

G

ivankravets, 2018-10-29:
> constantly moving carousel

Is it better if we disable it?

drgeoffb, 2018-10-30:
> > constantly moving carousel
> Is it better if we disable it?

By 'disable', do you mean make it go away altogether or disable the movement?

In either case, my answer is "Not necessarily, no".

I am easily distracted by shiny (and moving) things and it does affect my ability to concentrate on whatever I'm _supposed_ to be working on, but that's my problem; I'm not about to claim that it should be disabled for everyone just because it doesn't suit me. Folks probably appreciate the option to see recent announcements, PIO probably _want_ people to see recent announcements, and I'm sure the movement doesn't bother plenty of people.

I can see that the 'recent news' is cached, so I presume that it doesn't really generate very much 'useless traffic' at all, but if traffic is an issue for folks on satellite, dial-up or mobile links, perhaps disabling would be helpful for them.

I will appreciate the option to make it go away and/or just disable the constant movement just so that I don't have to tweak the CSS after updates, but it's really no biggie at all for me at this point. I've found a solution that removes the distraction for me, so I'm happy. You adding some options and/or disabling things would save me maybe thirty seconds a month so, really, don't make any particular effort on my account!

ivankravets, 2018-10-30:
We did a few things in the latest release of PIO Plus 1.5:

- removed "Recent" News" fetching on PIO Home start. Now, we check if there is new news, we load them IN BACKGROUND. I will not need to wait 5-10sec as before. Also, we cache previously fetched news. It should work fast now without generating extra traffic and extra using CPU
- stopped auto-playing of Recent News slides.

ivankravets, 2018-10-30:
Sure, we will add it soon. We plan to introduce "Settings" page for PIO Home. You will be able to control settings from PlatformIO Core too.

xMateMCPE, 2018-10-30:
Cool! Thanks for the fast reply.

drgeoffb, 2018-10-31:
This sounds great to me. Thanks, @ivankravets , for getting on to it so quickly. Best wishes!

borissprat, 2019-07-02:
Complete news blackout customisation has my vote - any small savings on my mobile link is appreciated and de-cluttering the home page likewise appreciated. 

I have copied the solution presented by fellow antipodean @drgeoffb which has de-cluttered the home page, but can someone please advise the best way to completely remove news fetching under atom on linux until a proper home page settings module is in place.

DarkAlchy, 2019-07-20:
We are now over half way through July of 2019 and still no ability to turn this screen space wasting stuff off?  Even my 1920x1200 is cluttered and I have to say this is the first time I have seen something like this without a settings/customization ability in it from the get go.  Bad, just bad form.