platformio/platformio-core #1892: Upload fails on jlink tool with spaces in path

cinderblock, 2018-10-24:
### Configuration

**Operating system**: Windows 10

**PlatformIO Version**: PlatformIO, version 3.6.1rc6

### Description of problem
Upload fails with jlink tool if there are spaces anywhere in the project's path.

#### Steps to Reproduce

1. Create new project that uses `tool-jlink`
   1. XMC1100 XMC2Go is an example "Board" that by default uses `tool-jlink`
   2. Save the project with a space somewhere in the path
      - Space in the project name: `Test Project`
      - Space in a parent directory of the project: `C:\My Projects\`
2. Click "Upload"

### Actual Results
Failure during upload:

```plain
[...]
Configuring upload protocol...
AVAILABLE: jlink
CURRENT: upload_protocol = jlink
Uploading .pioenvs\xmc1100_xmc2go\firmware.hex
SEGGER J-Link Commander V6.32h (Compiled Jul  5 2018 18:11:36)
DLL version V6.32h, compiled Jul  5 2018 18:11:19

Unknown command line option Project\.pioenvs\xmc1100_xmc2go\upload.jlink.
```

### Expected Results
`firmware.hex` is uploaded to device.

### Source and configuration

```platformio.ini
[env:xmc1100_xmc2go]
platform = infineonxmc
board = xmc1100_xmc2go
framework = arduino
```

```main.cpp
#include <Arduino.h>

void setup() {
    // put your setup code here, to run once:
}

void loop() {
    // put your main code here, to run repeatedly:
}
```

### Additional Notes

I could not for the life of me find where the PIO "tools" are defined so I'm posting this here for lack of a better location.

ivankravets, 2018-10-25:
Thanks a lot for the report. Next time please use https://github.com/Infineon/platformio-infineonxmc repository.

Fixed in https://github.com/Infineon/platformio-infineonxmc/releases/tag/v1.1.1. Please open PIO IDE > Left Sidebar > PlatformIO > Updates > Update All

closed 2018-10-25 by ivankravets

cinderblock, 2018-10-25:
@ivankravets Thank you for the link to the correct repo.

In the future, how do I go about finding a "map" of where tools are defined? To put it another way, which platformio controlled collection/list/page/database does pio lookup where `tools-jlink` is defined? This clearly exists for "[libraries](https://platformio.org/lib)", "[platforms](https://platformio.org/platforms)", "[frameworks](https://platformio.org/frameworks)", and "[boards](https://platformio.org/boards)". I just haven't found where the one for "tools" where I presume `tools-jlink` to be.

ivankravets, 2018-10-25:
Packages are here https://github.com/Infineon/platformio-infineonxmc/blob/master/platform.json#L16

cinderblock, 2018-10-25:
Thank you for the link.

After some more searching, I finally found a reference to the manifest file in the documentation: https://docs.platformio.org/en/latest/platforms/creating_platform.html

I'm surprised there isn't a first class ui for exploring "packages" like there is for other groupings of things:
[image]
[image]

ivankravets, 2018-10-26:
Do you mean this https://bintray.com/platformio?

cinderblock, 2018-10-27:
Hmmm. I think I understand why I was confused.

Originally, a couple months ago, I ran into this problem when I was in the earliest stages of creating my own framework.
I wanted to make a "tool" for Infineon's Memtool (which I use instead of jlink for xmc devices) and was looking to copy `tool-jlink` but could not find its definition. I had looked at the `infineonxmc` platform [packages page](https://platformio.org/platforms/infineonxmc/packages) but it only pointed to the official download locations of the tools and not to some platformio controlled url that I was expecting. This, among other factors, led me to put that effort on hold.

Then, when I ran into this issue, I jumped to the conclusion that `tool-jlink` was the culprit and did not fully investigate the platform.

Thank you very much for taking the time to help me find my way. 💯

Thank you for the awesome development environment!
I'm slowly moving all of my embedded stuff to PIO 😄