platformio/platformio-core #1902: Home: Could not initialize project

SkymanOne, 2018-10-29:
PIO Core Call Error: The next files/directories have been created in /Users/german_nikolishin/Projects/ArduinoBest
include - Put project header files here
lib - Put here project specific (private) libraries
src - Put project source files here
platformio.ini - Project Configuration File

Error: Processing uno (platform: atmelavr; board: uno; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/uno.html
PLATFORM: Atmel AVR > Arduino Uno
SYSTEM: ATMEGA328P 16MHz 2KB RAM (31.50KB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 24 compatible libraries
Scanning dependencies...
No dependencies
TypeError: extend() takes exactly one argument (2 given):
File "/Users/german_nikolishin/.platformio/penv/lib/python2.7/site-packages/platformio/builder/main.py", line 207:
json.dumps(env.DumpIDEData(), ensure_ascii=False)))
File "/Users/german_nikolishin/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/Users/german_nikolishin/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/pioide.py", line 145:
_dump_includes(env),
File "/Users/german_nikolishin/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/pioide.py", line 57:
env.subst("$PROJECTINCLUDE_DIR"), env.subst("$PROJECTSRC_DIR"))
========================== [ERROR] Took 0.41 seconds ==========================

pvbaudin, 2018-10-30:
I am having the same issue regardless of selected board profile.

lenisko, 2018-10-30:
Same problem, installed latest version (15min ago) of VSCode + PlatormIO, can't create new project.

ivankravets, 2018-10-30:
Please open PIO IDE Terminal and type `pio upgrade`. Sorry for the issue.

closed 2018-10-30 by ivankravets