platformio/platformio-core #1903: Create JSON schema for library.json

SRGDamia1, 2018-10-30:
What kind of issue is this?
- [x] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

It would be nice to have a JSON schema to validate library.json files against.

drorgl, 2022-11-25:
Try this:
https://gist.github.com/drorgl/28ab5cf4843acc61c33ac31146ed1102

Usage:
```
{
    "$schema": "https://gist.githubusercontent.com/drorgl/28ab5cf4843acc61c33ac31146ed1102/raw/70d687dbaddf9a631fdf538338787b9526380041/library_json_schema.json",
    "name": "test library",
    "build": {
        "srcDir": "./"
    }
}
```

ivankravets, 2022-11-25:
@drorgl, could you make PR to https://github.com/platformio/platformio-core/tree/develop/platformio/assets?

Please create a `schemas` folder and name the new schema as `library.json`. Thanks!

drorgl, 2022-11-27:
https://github.com/platformio/platformio-core/pull/4476

ivankravets, 2022-11-28:
Solved in https://docs.platformio.org/en/latest/manifests/library-json/fields/schema.html#manifest-library-json-schema

Thanks, @drorgl, for the PR #4476

commit 2141a097 (2022-11-28): A JSON Schema for library.json // Resolve #1903
closed 2022-11-28 by ivankravets