platformio/platformio-core #1904: Upload Auto-Detection finds the wrong port

prograMAnimus, 2018-10-30:
### Configuration

**Operating system**: Linux Arch

**PlatformIO Version**: 3.6.0

### Description of problem
I'm starting the Upload-procedure of my binary within the Eclipse-IDE.
My Application compiles just fine but when it tries to upload and detect the upload_port, it detects the wrong port, thus failing the upload.

Checking ``pio device list`` reveals that it has to be /dev/ttyACM3

Could it be due to the UDEV-Rules not detecting my Arduino?

### Actual Results
configuring upload protocol...
AVAILABLE: wiring
CURRENT: upload_protocol = wiring
Looking for upload port...
Auto-detected: /dev/ttyACM0

### Output of `pio device list`
```bash
/dev/ttyS0
----------
Hardware ID: n/a
Description: n/a

/dev/ttyACM3
------------
Hardware ID: USB VID:PID=2341:0042 SER=55736303631351816002 LOCATION=2-2.1:1.0
Description: Arduino Mega 2560

/dev/ttyACM2
------------
Hardware ID: USB VID:PID=0BDB:193E SER=1C131E4899FF65M0 LOCATION=2-4:1.9
Description: N5321 gw - N5321 gw Mobile Broadband GPS Port

/dev/ttyACM1
------------
Hardware ID: USB VID:PID=0BDB:193E SER=1C131E4899FF65M0 LOCATION=2-4:1.3
Description: N5321 gw - N5321 gw Mobile Broadband Data Modem

/dev/ttyACM0
------------
Hardware ID: USB VID:PID=0BDB:193E SER=1C131E4899FF65M0 LOCATION=2-4:1.1
Description: N5321 gw - N5321 gw Mobile Broadband Modem
```

### UDEV-Rules
```
# ....

# Arduino boards
SUBSYSTEMS=="usb", ATTRS{idVendor}=="2341", ATTRS{idProduct}=="[08][02]*", MODE:="0666"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="2a03", ATTRS{idProduct}=="[08][02]*", MODE:="0666"

# ....
```

### The content of `platformio.ini`:
```ini
[platformio]
env_default = megaatmega2560

[env:megaatmega2560]
platform = atmelavr
board = megaatmega2560
framework = arduino
build_flags = -D FirmwareVersion=0.1
lib_deps = Wire, OneWire
```

ivankravets, 2018-10-30:
Thanks! Fixed in https://github.com/platformio/platform-atmelavr/commit/8151ee29af271bfa8063b1c8ec22c3e254eadc0c

ivankravets, 2018-10-30:
Please re-test with development version https://github.com/platformio/platform-atmelavr#development-version

closed 2018-10-30 by ivankravets

prograMAnimus, 2018-10-31:
Thanks a bunch for the fast reply, but unfortunately it didn't seem to do the trick :(
I deleted the .pio*-Folders in the Project-Folder and Updated the platforms and libraries, so it could refresh the local files.

But trying to upload the .hex-File, I get the same error as before; auto-detect port fails.
I also ensured that the megaatmega2560.json file on my system is the same as the one in the dev-branch :(

### pio platform list
```
atmelavr ~ Atmel AVR
====================
Atmel AVR 8- and 32-bit MCUs deliver a unique combination of performance, power efficiency and design flexibility. Optimized to speed time to market-and easily adapt to new ones-they are based on the industrys most code-efficient architecture for C and assembly programming.

Home: http://platformio.org/platforms/atmelavr
Frameworks: arduino, simba
Packages: toolchain-atmelavr, framework-arduinoavr, framework-simba, tool-avrdude, tool-micronucleus
Version: 8151ee2
```

prograMAnimus, 2018-10-31:
Thanks a bunch! That actually did the trick!
/close