platformio/platformio-core #1905: PIO core not honoring src_filter directives

fiveangle, 2018-10-30:
### PIO core not honoring src_filter directives

On investigating a compile issue on the Marlin project (https://github.com/MarlinFirmware/Marlin/issues/12216), I discovered that pio core is not honoring the `src_filter ` directives, allowing LDF to pick up library inclusions from HAL platforms that it should never ever see for this `env`.

**Source:**
https://github.com/fiveangle/Marlin/tree/bf2bisect1

**PIO INI:**
https://github.com/fiveangle/Marlin/blob/bf2bisect1/platformio.ini

**Relevant env:**

```
[common]
default_src_filter = +<src/*> -<src/config> -<src/HAL> +<src/HAL/shared>

```
```
[env:at90usb1286_dfu]
platform      = teensy
framework     = arduino
board         = at90usb1286
build_flags   = ${common.build_flags}
lib_deps      = ${common.lib_deps}
lib_ldf_mode  = deep+
src_filter    = ${common.default_src_filter} +<src/HAL/HAL_AVR>
extra_scripts = pre:buildroot/share/atom/create_custom_upload_command_DFU.py
monitor_speed = 250000

```

Result is it attempts to compile `<Teensy_ADC>` library included in `src/HAL/HAL_LPC1768/HAL.h` which should have been excluded based on the INI `src_filter` rules that excludes the `src/HAL` tree.  I simplified the `src_filter` config for `env:at90usb1286_dfu` to omit the use of the "`common.default_src_filter`" so filters were listed in expected precidence order on single line within the `env`, but got identical results:

```
[ Deviot 2.3.0.dev7 ] Starting...

Processing at90usb1286_dfu (platform: teensy; board: at90usb1286; framework: arduino)

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/teensy/at90usb1286.html
PLATFORM: Teensy > Atmel AT90USB1286 based
SYSTEM: AT90USB1286 16MHz 8KB RAM (120KB Flash)
Converting Marlin.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(deep+) COMPATIBILITY(soft)
Collected 115 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4
|-- <LiquidCrystal> 1.3.4
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.2.2 #cefefbe
|   |-- <SPI> 1.0
|   |-- <SoftwareSerial> 1.0
|-- <Adafruit NeoPixel> 1.1.3
|-- <30aa480>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.6.1
|-- <c1921b4>
|   |-- <SPI> 1.0
|-- <Wire> 1.0
|-- <Servo> 1.1.2
|-- <Teensy_ADC>
|-- <SPI> 1.0
Compiling .pioenvs\at90usb1286_dfu\src\src\HAL\HAL_AVR\HAL.cpp.o
Compiling .pioenvs\at90usb1286_dfu\src\src\HAL\HAL_AVR\HAL_spi_AVR.cpp.o
Compiling .pioenvs\at90usb1286_dfu\src\src\HAL\HAL_AVR\MarlinSerial.cpp.o
[...]
Compiling .pioenvs\at90usb1286_dfu\lib783\ADC\ADC_Module.cpp.o
In file included from C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC.cpp:30:0:
C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC.h:41:2: error: #error "Sorry, this ADC library only works on ARM-based Teensy boards."
#error "Sorry, this ADC library only works on ARM-based Teensy boards."
^
In file included from C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC.h:49:0,
from C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC.cpp:30:
C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC_Module.h:48:2: error: #error "Board not supported!"
#error "Board not supported!"
^
In file included from C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC.h:49:0,
from C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC.cpp:30:
C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC_Module.h:336:2: error: #error "F_BUS must be 108, 60, 56, 54, 48, 40, 36, 24, 4 or 2 MHz"
#error "F_BUS must be 108, 60, 56, 54, 48, 40, 36, 24, 4 or 2 MHz"
^
C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC_Module.h:499:23: error: 'ADC_REFERENCE' has not been declared
void setReference(ADC_REFERENCE ref_type);
^
C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC_Module.h:764:20: error: 'ADC_INTERNAL_SOURCE' has not been declared
int analogRead(ADC_INTERNAL_SOURCE pin) __attribute__((always_inline)) {
^
compilation terminated due to -fmax-errors=5.
*** [.pioenvs\at90usb1286_dfu\lib783\ADC\ADC.cpp.o] Error 1
In file included from C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC_Module.cpp:32:0:
C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC_Module.h:48:2: error: #error "Board not supported!"
#error "Board not supported!"
^
In file included from C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC_Module.cpp:32:0:
C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC_Module.h:336:2: error: #error "F_BUS must be 108, 60, 56, 54, 48, 40, 36, 24, 4 or 2 MHz"
#error "F_BUS must be 108, 60, 56, 54, 48, 40, 36, 24, 4 or 2 MHz"
^
C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC_Module.h:499:23: error: 'ADC_REFERENCE' has not been declared
void setReference(ADC_REFERENCE ref_type);
^
C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC_Module.h:764:20: error: 'ADC_INTERNAL_SOURCE' has not been declared
int analogRead(ADC_INTERNAL_SOURCE pin) __attribute__((always_inline)) {
^
C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC_Module.h: In member function 'void ADC_Module::continuousMode()':
C:\Users\djohnson\.platformio\packages\framework-arduinoteensy\libraries\ADC\ADC_Module.h:636:41: error: 'setBit' was not declared in this scope
setBit(ADC_SC3, ADC_SC3_ADCO_BIT);
^
compilation terminated due to -fmax-errors=5.
*** [.pioenvs\at90usb1286_dfu\lib783\ADC\ADC_Module.cpp.o] Error 1
 [ERROR] Took 374.01 seconds
```

Thanks !

-=dave

------------------------------------------------------------------

### Configuration

Fails on both Windows 10 and Mac OS Mojave

$ pio --version 
PlatformIO, version 3.6.1rc6

ivankravets, 2018-11-19:
Thank you so much for the report! It's a very interesting bug which has been just fixed. Please re-test with the latest dev/version via `$ pio upgrade --dev`.

closed 2018-11-19 by ivankravets

fiveangle, 2018-11-19:
Works like a charm.

Thanks !

-=dave

hongquan, 2019-05-14:
Sadly, I upgraded to version 4.0.0a13, but still get this bug (not honor "-" in `src_filter`).

ivankravets, 2019-05-14:
Could you provide a simple project or configuration to reproduce this issue?

hongquan, 2019-05-15:
Hi @ivankravets 
Please try this source (please clone with `--recursive`). What I want to exclude is the path `lib/doser-arm-lib/lib/Unity`.

ivankravets, 2019-05-15:
`src_filter` works only for source code located in source folder. if you need to exclude something from a library, you need a custom `library.json` manifest with `srcFilter`. See http://docs.platformio.org/en/latest/librarymanager/config.html#srcfilter

hongquan, 2019-05-15:
Hi, I added a file _lib/doser-arm-lib/library.json_ with this content:

```json
"build": {
    "srcFilter": [
        "-<lib/Unity>"
    ]
}
```

but when building, I got this error:

```
Compiling .pioenvs/esp32vn-iot-uno/src/main.cpp.o
src/main.cpp:10:27: fatal error: doser-arm-lib.h: No such file or directory
```

hongquan, 2019-05-15:
Thanks @ivankravets 
Turn out that my _library.json_ file was wrong. It should be:

```json
{
    "name": "doser-arm-lib",
    "build": {
        "srcFilter": [
            "-<lib/Unity>"
        ]
    }
}
```

commit 7322df26 (2018-11-19): Fix an issue when Library Dependency Finder (LDF) does not handle project `src_filter ` // Resolve #1905