platformio/platformio-core #1912: Can't get platformio on raspberry, no output from cli.

gianiaz, 2018-11-06:
**PlatformIO Core**.
  I can't get any output from platformio cli in a raspberry installation. 

### Configuration
- Raspberry 2 Model B

pi@raspberrypi:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.4 (stretch)
Release:	9.4
Codename:	stretch

**PlatformIO Version** 

pi@raspberrypi:~ $ platformio --version
PlatformIO, version 3.6.1

### Description of problem

No output from platformio, on a working installation I have this output:

➜  ~ platformio account login -u gianiaz@gmail.com -p xxxxxxxxxx
***********************************************************************************************
Obsolete PIO Core v3.6.1 is used (previous was 3.6.2a2)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
***********************************************************************************************
PIO Plus (https://pioplus.com) v1.5.1
Successfully authorized!

➜  ~ platformio account show
***********************************************************************************************
Obsolete PIO Core v3.6.1 is used (previous was 3.6.2a2)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
***********************************************************************************************
PIO Plus (https://pioplus.com) v1.5.1

Logged as: gianiaz@gmail.com
PIO Plus Plan: PIO Plus Community

On raspberry:

pi@raspberrypi:~ $ platformio account login -u gianiaz@gmail.com -p xxxxxxxxx
pi@raspberrypi:~ $ platformio account show
pi@raspberrypi:~ $ 

The raspberry image is totally clean, I've downloaded the latest version and installed only platformio.

ivankravets, 2018-11-06:
Could you run the next commands on RPi side?
```
$ pio update
$ ls -al ~/.platformio/packages/tool-pioplus
$ ldd ~/.platformio/packages/tool-pioplus/pioplus
$ ~/.platformio/packages/tool-pioplus/pioplus
```

Please provide an output

gianiaz, 2018-11-06:
I've done a new installation from scratch, this is the entire log:

* VIRTUAL ENVIRONMENT *
pi@raspberrypi:~ $ virtualenv /home/pi/.platformio/penv
New python executable in /home/pi/.platformio/penv/bin/python
Installing setuptools, pip, wheel...
pi@raspberrypi:~ $ source .platformio/penv/bin/activate

* INSTALL *
(penv) pi@raspberrypi:~ $ pip install -U platformio
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting platformio
  Using cached https://files.pythonhosted.org/packages/15/f6/6a837ec8ebf25d8737cc72faa6fccc023b325f9a19003cb374e5323d730f/platformio-3.6.1-py27-none-any.whl
Collecting semantic-version<3,>=2.5.0 (from platformio)
Collecting requests<3,>=2.4.0 (from platformio)
  Using cached https://files.pythonhosted.org/packages/f1/ca/10332a30cb25b627192b4ea272c351bce3ca1091e541245cccbace6051d8/requests-2.20.0-py2.py3-none-any.whl
Collecting pyserial!=3.3,<4,>=3 (from platformio)
  Using cached https://files.pythonhosted.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl
Collecting click<6,>=5 (from platformio)
  Using cached https://files.pythonhosted.org/packages/8f/98/14966b6d772fd5fba1eb3bb34a62a7f736d609572493397cdc5715c14514/click-5.1-py2.py3-none-any.whl
Collecting colorama (from platformio)
  Using cached https://files.pythonhosted.org/packages/0a/93/6e8289231675d561d476d656c2ee3a868c1cca207e16c118d4503b25e2bf/colorama-0.4.0-py2.py3-none-any.whl
Collecting bottle<0.13 (from platformio)
Collecting urllib3<1.25,>=1.21.1 (from requests<3,>=2.4.0->platformio)
  Using cached https://files.pythonhosted.org/packages/62/00/ee1d7de624db8ba7090d1226aebefab96a2c71cd5cfa7629d6ad3f61b79e/urllib3-1.24.1-py2.py3-none-any.whl
Collecting idna<2.8,>=2.5 (from requests<3,>=2.4.0->platformio)
  Using cached https://files.pythonhosted.org/packages/4b/2a/0276479a4b3caeb8a8c1af2f8e4355746a97fab05a372e4a2c6a6b876165/idna-2.7-py2.py3-none-any.whl
Collecting chardet<3.1.0,>=3.0.2 (from requests<3,>=2.4.0->platformio)
  Using cached https://files.pythonhosted.org/packages/bc/a9/01ffebfb562e4274b6487b4bb1ddec7ca55ec7510b22e4c51f14098443b8/chardet-3.0.4-py2.py3-none-any.whl
Collecting certifi>=2017.4.17 (from requests<3,>=2.4.0->platformio)
  Using cached https://files.pythonhosted.org/packages/56/9d/1d02dd80bc4cd955f98980f28c5ee2200e1209292d5f9e9cc8d030d18655/certifi-2018.10.15-py2.py3-none-any.whl
Installing collected packages: semantic-version, urllib3, idna, chardet, certifi, requests, pyserial, click, colorama, bottle, platformio
Successfully installed bottle-0.12.13 certifi-2018.10.15 chardet-3.0.4 click-5.1 colorama-0.4.0 idna-2.7 platformio-3.6.1 pyserial-3.4 requests-2.20.0 semantic-version-2.6.0 urllib3-1.24.1
(penv) pi@raspberrypi:~ $ 

* FIRST LOGIN ATTEMPT *
(penv) pi@raspberrypi:~ $ platformio account login -u gianiaz@gmail.com -p MYSUPERSECRETPASS
CorePackageManager: Installing tool-pioplus @ ^1.5.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%
tool-pioplus @ 1.5.1 has been successfully installed!
CorePackageManager: Installing contrib-pysite @ >=0.3.2,<2
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
contrib-pysite @ 0.3.2 has been successfully installed!

* YOUR COMMANDS SUGGESTION *
(penv) pi@raspberrypi:~ $ pio update
Updating contrib-pysite                  @ 0.3.2          [Up-to-date]
Updating tool-pioplus                    @ 1.5.1          [Up-to-date]

(penv) pi@raspberrypi:~ $ ls -al ~/.platformio/packages/tool-pioplus
totale 2264
drwx------ 2 pi pi    4096 nov  6 17:54 .
drwxr-xr-x 4 pi pi    4096 nov  6 17:55 ..
-rw-r--r-- 1 pi pi     195 ott 29 16:44 package.json
-rwxr-xr-x 1 pi pi 2302567 ott 29 16:44 pioplus
(penv) pi@raspberrypi:~ $ ldd ~/.platformio/packages/tool-pioplus/pioplus
	non è un eseguibile dinamico (
not a dynamic executable*)
(penv) pi@raspberrypi:~ $  ~/.platformio/packages/tool-pioplus/pioplus
Istruzione non consentita (*Operation not allowed*)

ivankravets, 2018-11-06:
I've just booted my RPi 1 and have the same issues. I'll back here with news soon. Thank you so much that reported!

ivankravets, 2018-11-06:
Please run `pio update` again. It should work now.

closed 2018-11-06 by ivankravets