platformio/platformio-core #1913: Build a project with multiple configuration files "platformio.ini"

ivankravets, 2018-11-06:
Allow passing a custom "platformio.ini" or even a multiple list with configuration files. For example, `pio run -c my1.ini -c my2.ini`.

TD-er, 2018-12-03:
Could such ini files also be used to setup a personal config extension?
For example have a list of env's used in the platformio.ini and add the used com port or OTA settings for each of those.
These files are very specific for a build environment/PC, so no need to have it in your version control list.

ivankravets, 2018-12-14:
@TD-er it seems you mean this feature request https://github.com/platformio/platformio-core/issues/1590?

TD-er, 2018-12-14:
@ivankravets Looks like it indeed :)

closed 2019-05-09 by ivankravets

ivankravets, 2019-05-09:
Docs:
- http://docs.platformio.org/en/latest/userguide/cmd_run.html#cmdoption-platformio-run-c
- http://docs.platformio.org/en/latest/userguide/cmd_test.html#cmdoption-platformio-test-c
- http://docs.platformio.org/en/latest/userguide/cmd_debug.html#cmdoption-platformio-debug-c

commit c79b3ff7 (2019-05-09): Override default “platformio.ini” with a custom using "-c, --project-conf" option // Resolve #1913