platformio/platformio-core #1914: subfolder of lib mustn't be named *config*

atsteich, 2018-11-07:
- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Ubuntu 17.04 system**:

**3.6.2a2**

### I have the following folder structure, and everything is working fine. 
```
|-- Projectfolder
    |-- include
        |-- README
    |-- lib
        |-- lib1
            |-- src
                |-- lib1.h
                |-- lib1.cpp
        |-- lib2
            |-- src
                |-- lib2.h
                |--  lib2.cpp
    |-- src
        |-- main.cpp
    |-- platformio.ini
```
code of lib1.h: 
```void func1();```

code of lib1.cpp:
```
void func1(){ 
    //doanythinghere 
}
```

code of lib2.h:
`void func2();`

code of lib2.cpp:
```
#include <lib1.h>

void func2(){
    func1();
}
```

code of main.cpp
```
#include <lib2.h>

void setup(){
}

void loop(){
    func2();
}
```

BUT: As soon I am renaming  
```
lib1 -> config
lib1.h -> config.h
lib1.cpp -> config.cpp
```

I get an Compile Error:
lib/lib2/src/lib2.cpp:4:11: error: 'func1' was not declared in this scope

If for any reason this is not a bug but a feature, it should be mentioned in the Readme, i spent almost a day on finding out that the problem is the name of the lib...

**The content of `platformio.ini`:**
```
[env:featheresp32]
platform = espressif32
board = featheresp32
framework = arduino
```

ivankravets, 2018-11-10:
Did you rename `#include <lib1.h>` in `lib2.cpp:`?

atsteich, 2018-11-10:
> Did you rename `#include <lib1.h>` in `lib2.cpp:`?

Of course!

ivankravets, 2018-11-10:
Please provide archive with a simple project to reproduce this issue. Thanks!

atsteich, 2018-11-19:
has anybody been able to reproduce?

ivankravets, 2018-11-19:
Sorry for delay. Please upgrade PIO Core to the latest dev version via `pio upgrade --dev`. Does it work now?

ivankravets, 2018-11-19:
The issue was linked with duplicated `config.h` in Arduino framework which was in the first order in CPPPATH. Nevertheless, this is a bug, user's custom includes, project includes, and library dependencies should go first in list for C Preprocessor.

atsteich, 2018-11-19:
after upgrading everything works as expected, thanks!

nstiurca, 2019-03-20:
Sorry to resurrect an old issue, but it seems commit b22ca10 has broken my builds. I have a project where my `platformio.ini` has something like

```
[platformio]
include_dir = ./foo_lib
...

[env:...]
...
lib_deps =
  bar_lib
  ...
```

Prior to b22ca10, the build command would effectively end up with `-I./foo_lib` before `-I./.piolibdeps/bar_lib/` which worked. After this patch, the order is flipped, which I guess was the intent?

Unfortunately I have a `.h` file with the same name in both `foo_lib` and `bar_lib`, and it would be very difficult to resolve the name conflict cleanly. I need the version in `foo_lib`, not in the thirdparty `bar_lib` for compiling my code.

I'm a bit confused because if anything, the changes that were made to close this issue seem like they would have ensured that whatever I specified with `include_dir` would have ended up first, but this does not seem to be the case. Regardless, how do I ensure that `-I./foo_lib` comes early in the build flags?

ivankravets, 2019-03-21:
@nstiurca Please file a new issue here https://github.com/platformio/platformio-core/issues

ivankravets, 2019-03-23:
@nstiurca thank you so much for the report! Please re-test with `pio upgrade --dev`.

closed 2019-03-23 by ivankravets

nstiurca, 2019-03-25:
That fixed it for me, thanks for the quick turnaround!

commit b22ca10f (2018-11-19): Prepend CPPATH of library dependencies instead of appending // Resolve #1914
commit cae708f2 (2019-03-23): Fix an issue with incorrect order of project "include" and "src" paths in ``CPPPATH`` // Resolve #1914
commit 00b16260 (2019-03-23): Fix an issue with incorrect order of project "include" and "src" paths in ``CPPPATH`` // Resolve #1914