platformio/platformio-core #1921: Home: Could not initialize project

yusu24, 2018-11-11:
[Error 2] The system cannot find the file specified: u'C:\\Users\\M Yusuf Effendi\\Documents\\PlatformIO\\Projects'

ivankravets, 2018-11-19:
Do you have this folder `C:\Users\M Yusuf Effendi\Documents\PlatformIO\Projects`? Please note that you can set a custom path for projects. http://docs.platformio.org/en/latest/userguide/cmd_settings.html#projects-dir

closed 2018-11-19 by ivankravets

hansmbakker, 2018-12-02:
@ivankravets I have the same issue and see a lot of others having it, can it have to do something with permissions?

ivankravets, 2018-12-03:
@hansmbakker does your name also contain space project path?

hansmbakker, 2018-12-03:
No, my issue is python not behaving well with Controlled Folder Access in Windows (see https://github.com/platformio/platformio-core/issues/1952). So different problems can lead to the same symptom (the error message in the opening post)