platformio/platformio-core #1930: LDF chain+ and deep+ in atmelavr platform include spurious Robot_Control library

conklech, 2018-11-17:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

Windows 10

**PlatformIO Version** (`platformio --version`):

3.6.2a2, through VSCode plugin

### Description of problem

A minimal test case is linked below.

#### Steps to Reproduce

1. Create new project using atmelavr platform, arduino framework. I used megaatmega2560 board.
2. Add `lib_ldf_mode = chain+` or `lib_ldf_mode = deep+` to `platformio.ini`.
3. Add `#include <SD.h>` to main.cpp
4. Build.

### Actual Results

LDF erroneously determines that <SD> library depends on <Robot Control> library. Build fails because Robot Control cannot be built.

[Full output, with modified project files](https://gist.github.com/conklech/8c928810cc910c6b64a736c1cb9167a1)

Relevant portion of output:
```
Dependency Graph
|-- <SD> 1.1.1 (C:\Users\chris\.platformio\packages\framework-arduinoavr\libraries\SD)
|   |-- <Robot Control> 1.0.3 (C:\Users\chris\.platformio\packages\framework-arduinoavr\libraries\Robot_Control)
|   |   |-- <SPI> 1.0 (C:\Users\chris\.platformio\packages\framework-arduinoavr\libraries\__cores__\arduino\SPI)
|   |   |-- <Wire> 1.0 (C:\Users\chris\.platformio\packages\framework-arduinoavr\libraries\__cores__\arduino\Wire)
```
and
```
avr-g++ -o .pioenvs\megaatmega2560\libf55\Robot_Control\Arduino_LCD.cpp.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega2560 -DPLATFORMIO=30602 -DARDUINO_AVR_MEGA2560 -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO=10805 -IC:\Users\chris\.platformio\packages\framework-arduinoavr\cores\arduino -IC:\Users\chris\.platformio\packages\framework-arduinoavr\variants\mega -IC:\Users\chris\.platformio\packages\framework-arduinoavr\libraries\__cores__\arduino\SPI\src
-IC:\Users\chris\.platformio\packages\framework-arduinoavr\libraries\__cores__\arduino\Wire\src -IC:\Users\chris\.platformio\packages\framework-arduinoavr\libraries\Robot_Control\src C:\Users\chris\.platformio\packages\framework-arduinoavr\libraries\Robot_Control\src\Arduino_LCD.cpp
C:\Users\chris\.platformio\packages\framework-arduinoavr\libraries\Robot_Control\src\ArduinoRobot.cpp: In constructor 'RobotControl::RobotControl()':
C:\Users\chris\.platformio\packages\framework-arduinoavr\libraries\Robot_Control\src\ArduinoRobot.cpp:26:42: error: 'LCD_CS' was not declared in this scope
RobotControl::RobotControl():Arduino_LCD(LCD_CS,DC_LCD,RST_LCD){
^
```

leading to build failure.

### Expected Results

Build succeeds; LDF determines that library <SD> depends on library <SPI>, not on <Robot Control>.

```
Dependency Graph
|-- <SD> 1.1.1 (C:\Users\chris\.platformio\packages\framework-arduinoavr\libraries\SD)
|   |-- <SPI> 1.0 (C:\Users\chris\.platformio\packages\framework-arduinoavr\libraries\__cores__\arduino\SPI)
```

Changing to `chain` or `deep` LDF mode results in successful build, so this seems related to the CPP functionality implied by `+`.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:megaatmega2560]
platform = atmelavr
board = megaatmega2560
framework = arduino
lib_ldf_mode = deep+
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>
#include <SD.h> // Adding this include triggers inclusion of Robot_Control with deep+ library dependency resolution

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

### Additional info

conklech, 2018-11-17:
It looks like a similar or related bug was reported and fixed back in 2016: https://community.platformio.org/t/v3-framework-arduinoteensy-minimal-project-fails-to-build/548

ivankravets, 2018-11-19:
Thanks for the report! Please re-test with the latest development version via
```
$ pio upgrade --dev
```

conklech, 2018-11-19:
After updating with `pio upgrade --dev` (to reported version 3.6.2b2) I no longer get the spurious dependency, but get the following warnings for every file when building with verbose with either chain+ or deep+. It's not immediately obvious to me whether this is relevant/significant.

```
Warning! Classic Pre Processor is used for `****`, advanced has failed with `'tuple' object has no attribute 'add'`
```

If you'd like additional testing/investigation, let me know. Thanks!

conklech, 2018-11-20:
Running into a different problem now. [Here's a gist of the output, with the error case (deep+) as the second file and the succeeding case (deep) as the first.](https://gist.github.com/conklech/ee9c9714c2fd50fd31f6baa6d436f5ab). Same build inputs as in the original report, i.e. just `#include <SD.h>`.

Here it's compiling something that it should (`Sd2Card.cpp`, a dependency of the `SD` library), but it's not populating the include path correctly for some reason.

Thanks again for the continuing support!

ivankravets, 2018-11-20:
Wow! It's a very interesting bug in @scons. We have fixed in https://github.com/platformio/scons/commit/04bf778cb8749845cf1f21b6f06a742dac01b6ad

Please re-run `pio upgrade --dev` again.

closed 2018-11-20 by ivankravets

conklech, 2018-11-24:
Fix confirmed.

commit 23af9c90 (2018-11-19): Fix an issue when Library Dependency Finder (LDF) finds spurious dependencies in ``chain+`` and ``deep+`` modes // Resolve #1930
commit a2b0b289 (2018-11-20): LDF: Stop handling "define" and "undef" when condition fails; handle CPP files in "chain+" and "deep+" modes // Resolve #1930