platformio/platformio-core #1933: I can't start platformio home

davidleyabe, 2018-11-19:
Hi, 

I am trying to open platformio Home but I get this message:

> Start PIO Home Server
Error: Error: Traceback (most recent call last): File "c:.platformio\penv\lib\site-packages\platformio__main__.py", line 120, in main cli(None, None, None) File "c:.platformio\penv\lib\site-packages\click\core.py", line 700, in call return self.main(args, kwargs) File "c:.platformio\penv\lib\site-packages\click\core.py", line 680, in main rv = self.invoke(ctx) File "c:.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke return _process_result(sub_ctx.command.invoke(sub_ctx)) File "c:.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke return ctx.invoke(self.callback, ctx.params) File "c:.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke return callback(args, kwargs) File "c:.platformio\penv\lib\site-packages\platformio\commands\home.py", line 31, in cli pioplus_call(sys.argv[1:]) File "c:.platformio\penv\lib\site-packages\platformio\managers\core.py", line 132, in pioplus_call code = subprocess.call([pioplus_path] + args, kwargs) File "c:.platformio\penv\lib\subprocess.py", line 172, in call return Popen(*popenargs, **kwargs).wait() File "c:.platformio\penv\lib\subprocess.py", line 394, in init errread, errwrite) File "c:.platformio\penv\lib\subprocess.py", line 644, in _execute_child startupinfo) WindowsError: [Error 2] The system cannot find the file specified

[image]

I tried to run the command 

> pip install -U platformio

and everything seems to be ok
[image]

Could you please help me?

Thanks

ivankravets, 2018-11-19:
Please remove `C:/.platformio/penv` folder and restart IDE.

davidleyabe, 2018-11-19:
Hi Ivan, 

Thanks for your answer. I did the procedure that you mentioned but unfortunately the problem is not solved. I got a similar message. Could you please help me?

> Error: Error: Traceback (most recent call last): File "c:\python27\lib\site-packages\platformio__main__.py", line 120, in main cli(None, None, None) File "c:\python27\lib\site-packages\click\core.py", line 700, in call return self.main(args, kwargs) File "c:\python27\lib\site-packages\click\core.py", line 680, in main rv = self.invoke(ctx) File "c:\python27\lib\site-packages\click\core.py", line 1027, in invoke return _process_result(sub_ctx.command.invoke(sub_ctx)) File "c:\python27\lib\site-packages\click\core.py", line 873, in invoke return ctx.invoke(self.callback, ctx.params) File "c:\python27\lib\site-packages\click\core.py", line 508, in invoke return callback(args, kwargs) File "c:\python27\lib\site-packages\platformio\commands\home.py", line 31, in cli pioplus_call(sys.argv[1:]) File "c:\python27\lib\site-packages\platformio\managers\core.py", line 132, in pioplus_call code = subprocess.call([pioplus_path] + args, kwargs) File "c:\python27\lib\subprocess.py", line 172, in call return Popen(*popenargs, **kwargs).wait() File "c:\python27\lib\subprocess.py", line 394, in init errread, errwrite) File "c:\python27\lib\subprocess.py", line 644, in _execute_child startupinfo) WindowsError: [Error 2] The system cannot find the file specified

[image]

ivankravets, 2018-11-19:
Please open PIO IDE Terminal and type `pio home`. Do you see any errors?

davidleyabe, 2018-11-19:
I have the following error:

> C:\Users\Zôdio>pio home
***********************************************************************************************************************
Obsolete PIO Core v3.6.1 is used (previous was 3.6.2b1)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
***********************************************************************************************************************
Error: Traceback (most recent call last):
  File "c:\python27\lib\site-packages\platformio\__main__.py", line 120, in main
    cli(None, None, None)
  File "c:\python27\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\python27\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\python27\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\python27\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\python27\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\python27\lib\site-packages\platformio\commands\home.py", line 31, in cli
    pioplus_call(sys.argv[1:])
  File "c:\python27\lib\site-packages\platformio\managers\core.py", line 132, in pioplus_call
    code = subprocess.call([pioplus_path] + args, **kwargs)
  File "c:\python27\lib\subprocess.py", line 172, in call
    return Popen(*popenargs, **kwargs).wait()
  File "c:\python27\lib\subprocess.py", line 394, in __init__
    errread, errwrite)
  File "c:\python27\lib\subprocess.py", line 644, in _execute_child
    startupinfo)
WindowsError: [Error 2] The system cannot find the file specified

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

[image]

I am not able to solve it with the link that is specified. 

Do you have any other solutions?

Thanks

David

ivankravets, 2018-11-19:
Do you see `tool-pioplus` folder in `C:/.platformio/packages`? Please remove this folder and try again. Do you have an antivirus in a system?

davidleyabe, 2018-11-19:
Yes I have this folder and I removed it

[image]

But I have another issue:
[image]

My antivirus is AVAST.

ivankravets, 2018-11-19:
AVAST blocks PlatformIO. 
1. Please exclude `C:/.platformio` folder from AVAST
2. Please remove all files and folders in `C:/.platformio/`.
3. Restart IDE.

Everything should work now.

davidleyabe, 2018-11-19:
Hi Ivan, 
It works! Thanks for your help!
David

ivankravets, 2018-11-19:
Happy coding with PlatformIO!

closed 2018-11-19 by ivankravets