platformio/platformio-core #1951: Import of non-existent variable ''env'' using mbed framework

mupfelofen-de, 2018-12-02:
### Configuration
**Operating system**: `Linux 4.14.85 #1-NixOS SMP Sat Dec 1 08:43:00 UTC 2018 x86_64 GNU/Linux`

**PlatformIO Version** (`platformio --version`): `PlatformIO, version 3.5.3`

### Description of problem
`pio run` fails when trying to use the mbed framework.

#### Steps to Reproduce
1. Initialise new project: `pio init`
2. Modify `platformio.ini`:
```
[env:lpc1768]
platform = nxplpc
framework = mbed
board = lpc1768
```
3. Run it: `pio run -v`

### Actual Results
```
[Sun Dec  2 13:30:53 2018] Processing lpc1768 (platform: nxplpc; board: lpc1768; framework: mbed)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
PLATFORM: NXP LPC > NXP mbed LPC1768
SYSTEM: LPC1768 96MHz 64KB RAM (512KB Flash)
DEBUG: CURRENT(cmsis-dap) ON-BOARD(cmsis-dap) EXTERNAL(blackmagic, jlink)

*** Import of non-existent variable ''env''
File "/home/micha/.platformio/platforms/nxplpc/builder/frameworks/mbed.py", line 32, in <module>
```

makefu, 2018-12-02:
I can confirm this issue on NixOS. Maybe there is some environment library required for platformio implicit? Right now NixOS is bundling the following libraries: https://github.com/NixOS/nixpkgs/blob/master/pkgs/development/arduino/platformio/chrootenv.nix#L22-L30 

Cheers!

ivankravets, 2018-12-12:
> PlatformIO, version 3.5.3

Please upgrade to the latest PIO Core 3.6.

closed 2018-12-12 by ivankravets