platformio/platformio-core #1966: [platformio.ini] platform dependent bug with $PROJECTSRC_DIR

cyberman54, 2018-12-11:
Paths which use $PROJECTDIR in `platformio.ini` seems to run on platform dependent file not found error, as reported [here](https://github.com/cyberman54/ESP32-Paxcounter/issues/219):

File not found error with Ubuntu, but runs on Windows10:

```
build_flags_basic =
    -include $PROJECTSRC_DIR/hal/${PIOENV}.h
```
    
File not found error on Windows10, but runs on Ubuntu:

```
build_flags_basic =
    -include /$PROJECTSRC_DIR/hal/${PIOENV}.h
```

cyberman54, 2018-12-11:
Not a bug. Sorry, my fault. My path was invalid.

cyberman54, 2018-12-11:
I need to reopen this, because i closed it to soon. This issue persists.

ivankravets, 2018-12-12:
Have you resolved this issue?

cyberman54, 2018-12-12:
i found a workaround by wrapping the include parameters in "...", so can be closed, not sure if this is a clean solution, but seems to work.

ivankravets, 2018-12-12:
Great!

closed 2018-12-12 by ivankravets