platformio/platformio-core #1967: Add PLATFORMIO_ENV_DEFAULT environment variable

Niek, 2018-12-11:
There are [quite a few environment variables](https://docs.platformio.org/en/latest/envvars.html#general) that allow to override the platformio.ini options, but a variable for ``env_default`` is missing. It would come in handy to have a ``PLATFORMIO_ENV_DEFAULT`` (or similarly named) environment variable.

In case you're wondering why not simply use ``platformio run -e env``, this is not always possible. For example VSCode does not allow to specify arguments in the build task (environment variables are possible).

ivankravets, 2019-06-01:
Please re-test with the latest PIO Core 4.0 via `$ pio upgrade --dev`. Thanks!

closed 2019-06-01 by ivankravets

Niek, 2019-06-03:
Thanks!

I tested it, and it seems to run the build for both the ``env_default`` environment as well as the one provided with the ``PLATFORMIO_DEFAULT_ENVS`` variable. Not sure if that's intended, I think it would make more sense if the environment flag would overwrite the setting in the .ini file.

ivankravets, 2019-06-03:
Multiple option values have "Append" behavior for sys env. If you need to replace, then use in `platformIio.ini` => `default_envs = ${sysenv.PLATFORMIO...}

Niek, 2019-06-03:
OK, got it! Works fine otherwise

commit 23a2022f (2019-06-01): Add support for PLATFORMIO_DEFAULT_ENVS system environment variable // Resolve #1967