platformio/platformio-core #1969: Home: Libraries: Stats

GastonHP, 2018-12-13:
PIO Core Call Error: "<Exception instance at 0x53cb5c8 with str error:\n Traceback (most recent call last):\n  File \"C:\\Users\\GHP20\\.platformio\\packages\\contrib-pysite\\twisted\\python\\reflect.py\", line 448, in safe_str\n    return str(o)\nUnicodeEncodeError: 'ascii' codec can't encode character u'\\xe8' in position 1600: ordinal not in range(128)\n>"

ivankravets, 2018-12-13:
Could you reproduce this issue after VSCode restart?

GastonHP, 2018-12-13:
I get this kind of message since today.  And yes I already tried to restart, reinstall platform io, ...  This message also occurs with board list...

ivankravets, 2018-12-13:
What is your locale? Do you have non-ascii chars in path?

GastonHP, 2018-12-13:
But I still can build for my nodemcu and upload to it.

ivankravets, 2018-12-13:
For example, home folder. Does it contain non-ascii symbols? I'll try to reproduce this issue in a virtual machine.

GastonHP, 2018-12-13:
C:\Users\GHP20\.platformio
C:\Users\GHP20\Documents\PlatformIO\Projects\IREmitter
these are the paths where pio is installed and where my projects is saved.
maybe this is not pertinent: I tried today to use the esp8266FS to load FS on the nodemcu.  And this crashed.

GastonHP, 2018-12-13:
I use vscode on Windows 10 home.  And I did not change any folder name before I got this message.

GastonHP, 2018-12-13:
Other thing that changed since yesterday : 3.6.3 ???

ivankravets, 2018-12-13:
Please open PlatformIO IDE Terminal and type
```
pio lib stats --json-output
```

Do you see any warnings/errors?

GastonHP, 2018-12-13:
PS C:\Users\GHP20\Documents\PlatformIO\Projects\IREmitter> pio lib stats --json-output
{"updated": [{"date": "2018-12-13T11:31:03Z", "id": 2933, "name": "PushBullet-ESP8266"}, {"date": "2018-12-13T02:15:49Z", "id": 547, "name": "NeoPixelBus"}, {"date": "2018-12-13T00:00:27Z", "id": 13, "name": "Adafruit GFX Library"}, {"date": "2018-12-13T00:00:07Z", "id": 5958, "name": "ESPManager"}, {"date": "2018-12-12T23:50:06Z", "id": 5957, "name": "HT1621"}], "added": [{"date": "2018-12-13T00:00:07Z", "id": 5958, "name": "ESPManager"}, {"date": "2018-12-12T23:50:06Z", "id": 5957, "name": "HT1621"}, {"date": "2018-12-12T03:40:24Z", "id": 5956, "name": "EspMQTTClient"}, {"date": "2018-12-12T03:40:21Z", "id": 5955, "name": "SparkFun Qwiic
Twist Arduino Library"}, {"date": "2018-12-12T03:40:17Z", "id": 5954, "name": "WiFiMQTTManager Library"}], "dlweek": [{"diff": 378, "total": 1509, "id": 64, "name": "ArduinoJson"}, {"diff": 166, "total": 193, "id": 5513, "name": "TMCStepper"}, {"diff": 73, "total": 330, "id": 547, "name": "NeoPixelBus"}, {"diff": 39, "total": 905, "id": 89, "name": "PubSubClient"}, {"diff": 24, "total": 81, "id": 1647, "name": "WS2812FX"}, {"diff": 17, "total": 57, "id": 1269, "name": "painlessMesh"}, {"diff": 16, "total": 24, "id": 550, "name": "ThingSpeak"}, {"diff": 15, "total": 116, "id": 335, "name": "Brzo I2C"}, {"diff": 13, "total": 122, "id": 549, "name": "WebSockets"}, {"diff": 13, "total": 35, "id": 5774, "name": "MCCI LoRaWAN LMIC library"}], "topkeywords": ["communication", "display", "sensors", "control", "device", "lcd", "i2c", "other", "sensor", "input", "oled", "tft", "graphics", "signal", "output", "glcd", "spi", "font", "displaycore", "data", "serial", "led", "temperature", "timing", "arduino", "esp8266", "motor", "processing", "iot", "storage", "wifi", "accelerometer", "pwm", "rtc", "http", "uncategorized", "driver", "mbed", "mqtt", "esp32", "ethernet", "time", "humidity", "eeprom", "nucleo", "servo", "timer", "button", "adc", "web"], "dlmonth": [{"diff": 729, "total": 1964, "id": 13, "name": "Adafruit GFX Library"}, {"diff": 665, "total": 3967, "id": 89, "name": "PubSubClient"}, {"diff": 504, "total": 3036, "id": 28, "name": "Adafruit NeoPixel"}, {"diff": 448, "total": 5150, "id": 64, "name": "ArduinoJson"}, {"diff": 425, "total": 1471, "id": 31, "name": "Adafruit Unified Sensor"}, {"diff": 405, "total": 829, "id": 135, "name": "Adafruit SSD1306"}, {"diff": 378, "total": 1221, "id": 19, "name": "DHT sensor library"}, {"diff": 346, "total": 636, "id": 883, "name": "Servo"}, {"diff": 335, "total": 1270, "id": 567, "name": "WifiManager"}, {"diff": 294, "total": 1583, "id": 126, "name": "FastLED"}], "dlday": [{"diff": 10, "total": 10, "id": 1574, "name": "LiquidCrystal_I2C"}, {"diff": 6, "total": 6, "id": 2096, "name": "esphomelib"}, {"diff": 5, "total": 5, "id": 291, "name": "ClickEncoder"}, {"diff": 5, "total": 6, "id": 887, "name": "LiquidCrystal"}, {"diff": 4, "total": 5, "id": 38, "name": "Low-Power"}, {"diff": 4, "total": 5, "id": 258, "name": "TM1637"}, {"diff": 4, "total": 4, "id": 259, "name": "BLEPeripheral"}, {"diff": 4, "total": 4, "id": 560, "name": "Streaming"}, {"diff": 4, "total": 5, "id": 914, "name": "LedControl"}, {"diff": 4, "total": 5, "id": 1938, "name": "Ticker-esp32"}], "lastkeywords": ["uptime", "lamp", "lantern", "components", "dcf"]}

no error

ivankravets, 2018-12-13:
Please mail me@ikravets.com. I would be thankful for the remote session. Thanks!

GastonHP, 2018-12-13:
What kind of remote session do you ask?

ivankravets, 2018-12-13:
> Something strange : it works now!  No more messages!

Please reopen if you still have this issue.

closed 2018-12-13 by ivankravets

GastonHP, 2018-12-13:
Thanks for your quick answer.  I use your extension in vs code because it manages well my little Arduino projects as my IR emitter to control my old ir-devices thru google home.  I also tried visual micro and arduino ide.  But yours is my preferred one.

 

Regards,

 

Ghislain

 

L'absence de virus dans ce courrier électronique a été vérifiée par le logiciel antivirus Avast.
https://www.avast.com/antivirus

ivankravets, 2018-12-13:
Thank you so much for the kind words! Happy coding with PlatformIO! 😊