platformio/platformio-core #1977: Add lib_extra_dir support to library.json

ciband, 2018-12-15:
What kind of issue is this?
- [X] **Feature Request**.
      Due to this other issue (https://github.com/platformio/platform-espressif32/issues/148) I have to use the lib_ext_dir flag to pull in the correct library from the package.  Right now the only way to do this is to force users of my library to put this in their INI file.  I would like to put it in the JSON file so it is transparent to the users.

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 10 64

**PlatformIO Version** (`platformio --version`): 3.6.4b1

### Description of problem
lib_extra_dir cannot be defined in library.json

### Additional info
N/A

ivankravets, 2018-12-18:
I'm not sure that I understand you. Could you provide an example with `lib_extra_dir` + `library.json`?

ciband, 2018-12-18:
I'm not sure how.

I want to this lib_extra_dir flag as an option in the JSON file, similar to how the libLDFMode is exposed.

If this is not available then the user is forced to set it in their INI file to the correct value in order for my library to compile correctly.

I hope that makes sense.

ivankravets, 2018-12-26:
Does your library depend on other libraries? Could you provide a simple example to reproduce this issue?

closed 2019-04-18 by ivankravets