platformio/platformio-core #1983: Honor include/exclude when downloading a remote library

SRGDamia1, 2018-12-18:
What kind of issue is this?

- [x] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

------------------------------------------------------------------

The library.json file has include/exclude files for decreasing the library size in the archive.  It would be nice if it honored those when downloading a library directly from a git repo instead of the library archive.

I like to keep pdf's and other documentation in my git repo with my libraries.  When the library is zipped up and downloaded from the archive, they're correctly ignored.  But if someone wants to download the library from git, they have to wait while git processes and downloads all of those pdf's.

Note:  The presence/absence of these things makes no difference at all to the build.  This would just be to speed up downloads.

ivankravets, 2019-05-23:
Is it technically possible to clone Git repository and explain Git client which files to exclude? As I understand, Git clones the whole tree.

SRGDamia1, 2019-05-23:
You can use a sparse checkout:  https://git-scm.com/docs/git-read-tree

I admit, I don't know how to set it up right, though.

ivankravets, 2019-05-23:
This is tree information. We need something for `git clone`.

SRGDamia1, 2019-05-23:
I don't know of anything else in git.

ivankravets, 2019-05-23:
Sorry, I don't have an idea of how to resolve this issue in a simple form :(

closed 2019-05-23 by ivankravets